emsec/ChameleonMini #208: Mifare ultralight

ph553, 2019-02-25:
Hello
I was testing the mf ultralight emulation, 
The card is correctly detected on my android phone, with the right data but when i test it on the reader that the card i copied is used on, the reader doesn't detect it.
I set the green led on "field detected" and it lights up.
Also, if i try to upload data to the card from a bin on the iceman GUI, it fills it with dummy data (00000000 or ffffffffff) and the uid changes to 0000000.
I have to use teraterm to upload a dump correctly.
I was wondering if the first issue is maybe a timings problem? And if the second one is related to the first? Maybe i am missing a setting?

geo-rg, 2019-02-26:
Hi @ph553 
your problem with uploading a dump seems to me like a problem with the iceman GUI (@iceman1001 can you say something about this?), since it works properly with teraterm.

For the other problem, you could set `ledred=codec_rx`, which would indicate whether data is sent via NFC. Additionally, you could enable the log with `logmode=memory` and afterwards download the log with `logdownload`. **However, if you have a RevE rebooted Chameleon, I cannot help you here**, since we can only give you support for the RevG Chameleon by Kasper & Oswald.

iceman1001, 2019-02-26:
If you have a RevE rebooted the question is better asked  here ( https://github.com/iceman1001/ChameleonMini-rebooted )  or if its a GUI thing,  here ( https://github.com/iceman1001/ChameleonMini-rebootedGUI )

When you open an issue,  do please link your dump file and some screenshots. It will help out in the debugging process

ph553, 2019-02-26:
Hi
First of all thanks for your answer
I have both of the chameleons, however for this problem i am talking about the rev G
I will change the led setting and the logmode and keep you updated

iceman1001, 2019-02-26:
cool,  since you do have both devices,  start a thread in GUI repo.  It would be grate to have someone with a RevG doing some tests :)

ph553, 2019-02-26:
Alright, will do, let me just get home so i can post a proper issue with screenshots and all :)

ph553, 2019-02-26:
[log.txt](https://github.com/emsec/ChameleonMini/files/2906285/log.txt)
@geo-rg 
Here is the log file, the codex RX light was flashing about twice per second

iceman1001, 2019-02-26:
For the GUI part,  there has been a patch pushed.  The upload function didn't correctly detect if a dumpfile had a header field or not for MFU/NTAG in filetypes like BIN/EML ....

The rest of this issue is yours @geo-rg  :)

ph553, 2019-02-26:
@iceman1001 did you push it right now? If yes, that's impressive thanks

iceman1001, 2019-02-26:
yes,  its pushed.  Test and report back in GUI issue https://github.com/iceman1001/ChameleonMini-rebootedGUI/issues/100

ph553, 2019-02-26:
Okay, I will test tonight!

ph553, 2019-02-27:
@geo-rg it looks like in my log, the card data is not sent correctly to the reader, correctly if i'm wrong though, i am kind of new to reading ultralight data

geo-rg, 2019-02-27:
Hi @ph553 
the MF Ultralight emulation works as expected. On 10836 ms (and on other occasions), the reader requests a read on page address 0x11, but a MF Ultralight only has 16 pages (maximum page address 0x0F). I suspect that your reader expects a MF Ultralight EV1 or MF Ultralight C which have a bigger memory. You could simply use a MF Ultralight EV1 config on the Chameleon.

ph553, 2019-02-27:
Hey @geo-rg
I just finished testing it and it worked! Hooray!
On the revG, emulation works on any reader, while on the rev E rebooted, emulation was only working on some readers.
Thank you for the help, however i noticed another issue with the "readonly" function, but i will be closing this issue and opening another one in order to keep it clean. 
Thanks

closed 2019-02-27 by ph553