emsec/ChameleonMini #210: Readonly function

ph553, 2019-02-27:
Hi
I noticed the readonly sets itself to 0 on boot or on setting change
I checked the files and saw that on Configuration.c
#ifdef CONFIG_MF_ULTRALIGHT_SUPPORT
   [code.....]
        .ReadOnly = false
    }
Is it something i need to change in a file before compiling? I would like to get the readonly permanent and, if possible, not be a global variable

ph553, 2019-02-27:
@iceman1001 i noticed the same issue on the rev E rebooted :)

geo-rg, 2019-02-28:
Hi @ph553 
that's correct. Sadly, this cannot be fixed easily because of the current behavior of configs and settings. Only the settings are stored and recalled into the eeprom (and the memory content of each setting is stored to/recalled from the Flash) - the configurations, however, not, since they only contain function pointers, the memory size and the readonly flag. 

Thus, the only possible fix would be to move the flag from config to setting. In this case, however, the flag would be a setting property and would stay even if a new config is set for this setting (if not for an implementation of a special case).

geo-rg, 2019-02-28:
I changed the title and added the enhancement label since this is more an enhancement request. :)

ph553, 2019-02-28:
Ahh.. i had hoped it was an easy fix so i could do it myself.. too bad
Is it possible to default it to true instead?
(We could add a if setting=1, readonly = true)
Or maybe i could keep the chameleon ON all the time, does it drain that much battery?

geo-rg, 2019-02-28:
The chameleon in emulation mode should not drain so much battery. The LED setting `field_detected` triggers an ADC measurement every ~100 ms, this may increase the power consumption.

Please note that the chameleon has NO module that checks the battery voltage. Thus, if the battery is low, it may be exhausted quickly.

To set readonly to true per default, you could simply set the readonly flags in [Configuration.c](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Configuration.c) to true.

ph553, 2019-02-28:
Yup, as i suspected for the configurations.c, thanks
Should I close the issue or keep it open?

geo-rg, 2019-02-28:
Keep it open so we can close it, when this feature is added reliably.