emsec/ChameleonMini #211: Reader attack

ictmehrdad, 2019-03-01:
1- How have a reader attack with the tool!?
2- How can we crack the Code that we sniffed !?

ceres-c, 2019-03-01:
I have already replied you with all the needed information here: https://github.com/emsec/ChameleonMini/issues/203#issuecomment-458690224

Opening other issues with old questions won't lead to new answers because [you were already told](https://github.com/emsec/ChameleonMini/issues/203#issuecomment-458520065) we can't give you any step by step advice.
Please stop asking the same question and read technical documents.

geo-rg, 2019-03-01:
Hi @ictmehrdad 
as [I commented](https://github.com/emsec/ChameleonMini/issues/203#issuecomment-467375025) on your other issue: please open a new issue only if you have a **new question** that is not related to your old issue. This is not the case here.
You may re-open your old issue, if you have a **related** new question, which is also not the case since you re-asked your old question.

closed 2019-03-01 by geo-rg