emsec/ChameleonMini #213: Reader distance

ph553, 2019-03-05:
Hello again
I noticed mifare ultralight emulation(i dont know about lifare classic yet since i have nothing to test it with) does not work very well when the chameleon is too close to the reader. I read in other issues that the reader has to be at a 1cm distance from the chameleon.
I would like to know 
1. Is there something on my end i could do to make it more reliable at shorter distances
2. Won't that be a problem for reader that has a "slot-in" type where you have to insert the chameleon in a slot for it to be read?
Again thank you and I am sorry if I ask alot of questions

geo-rg, 2019-03-08:
Hi @ph553 
first of all: there's no need to apologize for asking questions :)

1. We do not know what causes this problem. So the only thing you could do would be testing and providing more information about in which cases (which reader, different positions, etc.) this happens.
2. Every reader is different. We never had problems with this type of reader.

ph553, 2019-03-08:
So i have the 3d printed case, however i suspect the 3d printer at my school doesn't work well since the chameleon didn't fit perfectly in the case: only the bottom side(where there the case isn't flat) fits.
When i put the chameleon on the reader with no distance between both, nothing happens