emsec/ChameleonMini #214: Different Communication on two same-type readers

ph553, 2019-03-08:
Hi
So basically I tried testing the chameleon on the local transportation system.
I have no problem in the metro, as long as i keep the chameleon at a certain distance, it's recognized as a valid pass, emits a green light and lets you in. This is the communications log from that transaction:
[workingLog.txt](https://github.com/emsec/ChameleonMini/files/2944165/workingLog.txt)

However, when i try it on the bus, i have no reaction. The readers are not exactly the same in appearance, but the communications should be the same.
This is what i have:
[notWorkingLog.txt](https://github.com/emsec/ChameleonMini/files/2944169/notWorkingLog.txt)

As you can see, it asks for multiple page reads on the first log, whereas it only asks for it once on the second.
Those are the logs from a rev G.

The bus reads the rev E rebooted fine on the original firmware, while only a few of the metro readers recognize it.

geo-rg, 2019-03-08:
Hi @ph553 
in your notWorkingLog, your emulation answers to the GetVersion command with the same bytes like in the other log. However, after this, the reader stops communication and re-tries. Then, when the reader sends the GetVersion the next time and gets the same answer, the communication stops, eventually. 

I suspect that the reader in the bus expects another GetVersion response. You could check this by checking the revE rebooted log from the bus.

ph553, 2019-03-08:
@geo-rg
The metro and the bus accept the same card, why would the version not be accepted?

I didn't know the rev E had a log, i will check it and come back with new information, thanks!

geo-rg, 2019-03-08:
@ph553 You're right, this is very strange. The Chameleon replies with the exact same response to the reader and communication works perfectly fine until the GetVersion command.

Is it possible that the Chameleon had a low battery in the bus? It would be a ridiculous explanation, but it is the only possible one that comes into my mind.

ph553, 2019-03-08:
I'll charge it and try again, however i doubt that's the problem since the rev G is new(does it ship with full battery?) and the metro still works even after the bus failed
Maybe it's because mfu ev1, as suggested in #208?
When i read the card from which i cloned it, it's a mifare 16 bytes regular card.. but then if I use regular mifare it doesn't work in the metro nor in the bus.. checkmate

ph553, 2019-03-08:
@geo-rg 
I just finished sniffing the communication between the rev E rebooted and the reader,
this is what i get:
[sniffLog.txt](https://github.com/emsec/ChameleonMini/files/2947088/sniffLog.txt)
This is the same card dump but the communication is quite different However i cannot distinguish between RX and TX

ph553, 2019-03-14:
@geo-rg I sniffed the comms from a real ticket, this is what i get : 
[comms.txt](https://github.com/emsec/ChameleonMini/files/2967840/comms.txt)

from this dump : 

[2tino2.zip](https://github.com/emsec/ChameleonMini/files/2967848/2tino2.zip)

Is the card answering with a 0504 ATQA at the beginning?
I see that the reader is not asking for a read on page 11 neither

I could try adding code to the Mifare Ultralight.C, if you have some clues to give me, it would be appreciated :)

geo-rg, 2019-03-20:
@ph553 
OK, I also don't understand the output from the RevE rebooted. Maybe @iceman1001 can help here?

The sniff log is very interesting. I think it did not understand the communication in the beginning correctly and thus, the ATQA is not visible. Generally, the card responses are not recorded very well by the Chameleon. 

However, starting from 30439 ms, the reader reads all pages from page 0 until page 15 in a very dumb way, because a MF Ultralight always responds with **four** pages to the read command. The reader in this log first reads page 0 (response is page 0,1,2,3), then page 1 (response is 1,2,3,4) and so on and thus reads every page multiple times. Another possibility is that this is no Mifare Ultralight card. Have you tried to `identify` it? What is the ATQA and SAK?

After reading out all these pages, the reader then writes the pages C, D, E, F, 4 and 3, which is another thing that the reader did not even try in the Chameleon logs you posted before. 

In the end, I think I cannot help you analyzing the system. From what I can see, the Chameleon works as it should, so please consider closing this issue. This issue board is not for help in analyzing systems (as much as I'd appreciate this :D ) but for help with the Chameleon.

iceman1001, 2019-03-20:
The reading of the data is interesting,  since as @geo-rg  states UL responses with 16bytes
Normally you can see after each block read that the data has shifted 4 steps.
But for  0x0B  and 0x0C block reads the data doesn't shift 4bytes, but looks quite wrong. 
Don't know the quality of sniffing you did or how,  and as @geo-rg  says this is not the place for such discussions.  

Try finding a chameleon mini forum,  or libnfc forum,  or pm3 forum to ask questions.

ph553, 2019-03-20:
Alright thanks for your help, i will be closing this issue!

closed 2019-03-20 by ph553