emsec/ChameleonMini #215: bootloader mode - test FAILED

banannd, 2019-03-19:
HI,
I ran the upgrade command and now I am in the bootloader mode (I did not update firmware). Now when I run the device I have a test failed
showlog
Generating Test Report from SRAM:
Test Report for Chameleon-Mini
* FW version: 0.5
* Test      Buttons: FAILED (Took 16539ms) Time exceeded without input on Left Button, time exceeded without input on Right Button.
* Test       Reader: FAILED (Took 521ms) Could not perform a whole ISO14443A select protocol and finally receive a SAK without errors. Used threshold range: 50 - 4095.
* Test         FRAM: PASSED (Took 143ms) Wrote 32768 pseudorandom bytes to FRAM. Successfully read the same bytes from FRAM.
* Test        Fuses: PASSED (Took 1ms) Read the following fuse bytes (the values in brackets have the form <reference value;mask>). 1: 00(00;FF) - 2: BE(BE;63) - 4: F3(F3;1F) - 5: ED(ED;3F).
* Test    Signature: PASSED (Took 0ms) Read the Device ID 1E9746, expected 1E9746.
* Test     USB Enum: PASSED (Took 120ms) USB Device State is ADDRESSED.
* Test          VCC: PASSED (Took 103ms) Measured a VCC of 3183 mV.
* Test      Antenna: PASSED (Took 104ms) Measured the following RSSI values: 11542 mV, 11625 mV, 11566 mV, 11601 mV, 11660 mV.
* Total Testing Time: 17533ms
Your Chameleon will emulate a card with the UID 8445C51B.
Sorry. Some test(s) have FAILED, please unplug chameleon, fix errors and re-test.
^Croot@nano-LIFEBOOK-E744:~# ^C

geo-rg, 2019-03-19:
Hi @banannd 
I assume that before running the upgrade command, the terminal output said something like "all tests have been passed SUCCESSFULLY", right? If so, there is no problem, because your test failed because of two tests that need user interaction. **However, there is no need to restart the test procedure.**

I assume you also typed `reset`, which resets the selftest procedure? This would explain the behavior. 

When you are in bootloader mode (directly after typing `upgrage<enter>`), you have to upgrade the firmware according to [the Getting Started](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__getting_started.html) -- please start with Step 2 since you are already in bootloader mode.

banannd, 2019-03-19:
Yes before  Irunning the upgrade command, the terminal output said something like "all tests have been passed SUCCESSFULLY"
and next I also typed reset.

do I have now to upgrade the firmware according ?
or can I abort the procedure and use the device
my FW version: is 0.5
when i now enter the command UID? or VERSION? or HELP system still show: 
Welcome to your ChameleonMini! This is test firmware version 0.5.
Sorry. Some test(s) have FAILED, please unplug chameleon, fix errors and re-test.
Type "upgrade" to jump to the bootloader and install new firmware as described on https://github.com/emsec/chameleonmini 
 
You can also use the following commands:
* Type "showlog" to show the factory test log
* Type "reset" to re-run the self test (you need to power cycle the Chameleon)

geo-rg, 2019-03-20:
Hi @banannd 
the Chameleon comes with a pre-installed test firmware which is basically useless for you. It has the purpose that you can see that the device did work as expected after production.

You need to upgrade the firmware as I wrote in my last comment.

banannd, 2019-03-20:
Thank you very much.

closed 2019-03-20 by geo-rg