emsec/ChameleonMini #217: start with chamlog.py - problem

banannd, 2019-04-04:
HI,
I'm starting with chamlog.py I downloaded a file from
https://github.com/emsec/ChameleonMini/tree/master/Software
I run the command ~ / chamlog $ sudo python3 chamlog.py -f / chamlog
and gets a message
   File "chamlog.py", line 109
     main ()
        ^
IndentationError: expected an indented block
the same at
  python3 chamtool.py -p / dev / ttyACM0 -v
   File "chamtool.py", line 284
     main ()
        ^
IndentationError: expected an indented block
How can I do it?

gtpy, 2019-04-05:
Your command is wrong.
You need to set the full path to the logfile like: `python chamlog.py -f ~/way/to/logfile/YourLogfile`
for example: `python chamlog.py -f ~/git/ChameleonMini/Software/logcham`

Or if you have the logfile and chamlog.py in same folder: 
`python chamlog.py -f YourLogfile`

banannd, 2019-04-08:
I have the same all the time
do I have to run a command on chameleon mini early?

~/chamlog$ sudo python3 chamlog.py -f logfile 
  File "chamlog.py", line 109
    main()
       ^
~/chamlog$ sudo python3 chamlog.py -f /chamlog/logfile   
File "chamlog.py", line 109
    main()
       ^

geo-rg, 2019-04-09:
Hi @banannd 
please run `git pull` in your local repo folder. If everything is up-to-date, the script should run. You could also check manually whether your chamlog.py matches the one that is uploaded here.

banannd, 2019-04-18:
Ok I've resolved this

closed 2019-04-18 by banannd