emsec/ChameleonMini #220: Sniff Both Way problem

banannd, 2019-04-18:
HI,
I have uploaded Firmware from #180 and in my log i have only RX transmission.
00000 ms <    +0 ms>:BOOT                         (0   bytes)	[]
26881 ms <+26881 ms>:CODEC RX SNI READER          (1   bytes)	[52]
27401 ms <  +520 ms>:CODEC RX SNI READER          (1   bytes)	[52]
27917 ms <  +516 ms>:CODEC RX SNI READER          (1   bytes)	[52]
28437 ms <  +520 ms>:CODEC RX SNI READER          (1   bytes)	[52]
28953 ms <  +516 ms>:CODEC RX SNI READER          (1   bytes)	[52]
29475 ms <  +522 ms>:CODEC RX SNI READER          (1   bytes)	[52]
29993 ms <  +518 ms>:CODEC RX SNI READER          (1   bytes)	[52]
31050 ms < +1057 ms>:CODEC RX SNI READER          (1   bytes)	[52]
31567 ms <  +517 ms>:CODEC RX SNI READER          (1   bytes)	[52]
32088 ms <  +521 ms>:CODEC RX SNI READER          (1   bytes)	[52]
32609 ms <  +521 ms>:CODEC RX SNI READER          (1   bytes)	[52]
what should I do to read TX transmission ?

banannd, 2019-04-25:
Ok I've resolved this.

closed 2019-04-25 by banannd

gtpy, 2019-04-25:
@banannd how you resolve it?
Remember - sharing is caring. Bring something back to the community. ;)

banannd, 2019-04-25:
I set slot 1 like as mifare classic card, and then i have in log TX and RX CODEC.

sp4rrows, 2019-05-17:
Hi I get the same issue.

MF Classic Card 1k
Smartphone nfc Tag Info as Reader
ChameleonMini 
slot1: 
config= 
Slot2:
configure=ISO14443A_SNIFF

Slot 2 active. 

Hold between reader and tag.

Minicom:
Logdownload 

Cham log: Python chamlog.py -f 

Could only get RX and no TX.
Setting slot1 to mifare classic don’t solve the problem.
Has anyone a idea how it works better.

fptrs, 2019-05-17:
Hi @sp4rrows,
you need to use the 'autocalibrate' cmd to calibrate the ISO14443A_SNIFF application.
So launch the cmd and hold the ChameleonMini between reader and tag. After a successful calibration the sniffer also receives transmissions from PICC to PCD.

sp4rrows, 2019-05-18:
hi @fptrs I tried it, but autocalibrate only works with the iso14443a_reader mode not with the sniff mode.
Didn't get the TX signals from PICC

banannd, 2019-05-20:
HI,
try upgrade firmware from:
https://github.com/gypsophlia/ChameleonMini
i test this on physical reader.

sp4rrows, 2019-05-20:
> HI,
> try upgrade firmware from:
> https://github.com/gypsophlia/ChameleonMini
> i test this on physical reader.

Hi @banannd 
I tried the firmware from @gypsophlia too.
Used master branch, but no success with my android smartphone and acr122u as reader to get the TX.

<details><summary> Card:</summary>
<p>
[usb] pm3 --> hf search
[=] Checking for known tags...
          
 UID : 2D 63 32 49           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1          
[=] proprietary non iso14443-4 card found, RATS not supported          
[=] Answers to magic commands: NO           
[+] Prng detection: HARD           
          
[+] Valid ISO14443-A tag  found
</p>
</details>

<details><summary> PM3 Sniff:</summary>
<p>

[usb] pm3 --> hf 14a sniff
#db# Starting to sniff          
#db# maxDataLen=3, Uart.state=0, Uart.len=0          
#db# traceLen=2743, Uart.output[0]=00000026          
[usb] pm3 --> hf list
[+] Recorded Activity (TraceLen = 2743 bytes)          
[=]           
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
iso14443a - All times are in carrier periods (1/13.56Mhz)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |       1056 | Rdr |26                                                                       |     | REQA          
      75904 |      76960 | Rdr |26                                                                       |     | REQA          
     574704 |     575760 | Rdr |26                                                                       |     | REQA          
     650352 |     651408 | Rdr |26                                                                       |     | REQA          
    4206928 |    4207984 | Rdr |26                                                                       |     | REQA          
    4282960 |    4284016 | Rdr |26                                                                       |     | REQA          
    4781776 |    4782832 | Rdr |26                                                                       |     | REQA          
    4857424 |    4858480 | Rdr |26                                                                       |     | REQA          
    8414000 |    8415056 | Rdr |26                                                                       |     | REQA          
    8489904 |    8490960 | Rdr |26                                                                       |     | REQA          
    8988704 |    8989760 | Rdr |26                                                                       |     | REQA          
    9064352 |    9065408 | Rdr |26                                                                       |     | REQA          
   12620928 |   12621984 | Rdr |26                                                                       |     | REQA          
   12696832 |   12697888 | Rdr |26                                                                       |     | REQA          
   13195648 |   13196704 | Rdr |26                                                                       |     | REQA          
   13197892 |   13200260 | Tag |04  00                                                                   |     |           
   13216240 |   13216656 | Rdr |01                                                                       |     |           
   13220592 |   13221008 | Rdr |01                                                                       |     |           
   13229296 |   13230480 | Rdr |00!                                                                      |     |           
   13233648 |   13234832 | Rdr |00!                                                                      |     |           
   13238000 |   13239184 | Rdr |2d!                                                                      |     |           
   13242352 |   13243536 | Rdr |58                                                                       |     |           
   13247088 |   13248208 | Rdr |b6                                                                       |     |           
   13946224 |   13947280 | Rdr |26                                                                       |     | REQA          
   13948468 |   13950836 | Tag |04  00                                                                   |     |           
   13962592 |   13963584 | Rdr |78!                                                                      |     |           
   13966816 |   13968000 | Rdr |01                                                                       |     |           
   13971168 |   13972352 | Rdr |01                                                                       |     |           
   13975520 |   13976704 | Rdr |00!                                                                      |     |           
[... 111 lines cut ...]
63053 ms <    +2 ms>:CODEC RX         (9   bytes) [93702d6332493516bd]
63073 ms <   +20 ms>:CODEC RX         (4   bytes) [500057cd]
63078 ms <    +5 ms>:CODEC RX         (1   bytes) [52]
63080 ms <    +2 ms>:CODEC RX         (9   bytes) [93702d6332493516bd]
63091 ms <   +11 ms>:CODEC RX         (4   bytes) [500057cd]
[... cut ...]
63120 ms <    +4 ms>:CODEC RX         (4   bytes) [500057cd]
63123 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
63124 ms <    +1 ms>:CODEC RX         (9   bytes) [93702d6332493516bd]
63128 ms <    +4 ms>:CODEC RX         (4   bytes) [6000f57b]
63130 ms <    +2 ms>:CODEC RX         (8   bytes) [8c576eaa295edb7d]
63133 ms <    +3 ms>:CODEC RX         (4   bytes) [500057cd]
63137 ms <    +4 ms>:CODEC RX         (1   bytes) [52]
...

Couldn't find a reason why it doesn't work with smartphone or ACR.
I'll try it next day on real readers.

gypsophlia, 2019-05-22:
It can do auto calibrate in sniffing mode, the steps are:
1. Reset threshold to the default value
2. put the ChameleonMini between the reader and the card
3. Use the following script for sending anticollision commands continuously, this is required
since the implementation of sniffing function tracks the anticollision process.

``` bash
while true
do
nfc−list −t 1
sleep 0.5
done
```

4. Issue command “timeout=600” and “autocalibrate” to ChameleonMini for extending the
timeout and executing autocalibrate function.

5. Check the response in the ChameleonMini terminal to determine if the autocalibrate have
found a threshold that works for sniffing.

6. After the autocalibrate is success, issue “nfc-list -t 1” command to the reader to see if
ChameleonMini can successfully sniff the communication under the threshold set by
autocalibrate

Also, you could try different thresholds manually. 

And I think you should use either SniffBothWay14443-pr branch of gypsophlia/ChameleonMini or master branch of emsec/ChameleonMini as it has been merged to the master branch of the original repo.  But I'm not sure if there are any commit that breaks the sniffing function after the merge.

sp4rrows, 2019-05-26:
@gypsophlia 
I tried your steps.
autocalibrate return 512
Used nfc-list -t 1 and sniff the traffic.

Logdownload:
python chamlog.py -f abc

Note: If parityBit check failed, '!' is appended to the decoded data and raw data with parity bit is displayed.

Traceback (most recent call last):
  File "chamlog.py", line 109, in <module>
    main()
  File "chamlog.py", line 102, in main
    log = Chameleon.Log.parseBinary(handle, args.decode)
  File "/home/git/ChameleonMini_upstream/ChameleonMini/Software/Chameleon/Log.py", line 130, in parseBinary
    logData = eventTypes[event]['decoder'](logData)
  File "/home/git/ChameleonMini_upstream/ChameleonMini/Software/Chameleon/Log.py", line 51, in binaryParityDecoder
    isValid, checkedData = checkParityBit(data)
  File "/home/git/ChameleonMini_upstream/ChameleonMini/Software/Chameleon/Log.py", line 23, in checkParityBit
    bit = (data[byteIndex] >> bitIndex) & 0x01
TypeError: string indices must be integers, not float

Will try it even more times but I don't understand why it not work every time.

sp4rrows, 2019-05-26:
Ok was my fault. used python 2.7

With python3 it looks like this:
python3 chamlog.py -f abc

Note: If parityBit check failed, '!' is appended to the decoded data and raw data with parity bit is displayed.

17982 ms <+17982 ms>:CODEC RX SNI READER          (1   bytes) [26                  ]    
17982 ms <    +0 ms>:CODEC RX SNI CARD W/PARITY   (3   bytes) [0400                ]    
17983 ms <    +1 ms>:CODEC RX SNI READER          (2   bytes) [9320                ]    
17984 ms <    +1 ms>:CODEC RX SNI CARD W/PARITY   (6   bytes) [1122334444          ]    
17986 ms <    +2 ms>:CODEC RX SNI READER          (9   bytes) [93701122334444519c  ]    
17986 ms <    +0 ms>:CODEC RX SNI CARD W/PARITY   (4   bytes) [08b6dd              ]    
17996 ms <   +10 ms>:CODEC RX SNI READER          (4   bytes) [500057cd            ]    
18007 ms <   +11 ms>:CODEC RX SNI READER          (1   bytes) [26                  ]    
18012 ms <    +5 ms>:CODEC RX SNI READER          (1   bytes) [26                  ]    
18018 ms <    +6 ms>:CODEC RX SNI READER          (1   bytes) [26                  ]    
19381 ms < +1363 ms>:CODEC RX SNI READER          (1   bytes) [26                  ]    
19381 ms <    +0 ms>:CODEC RX SNI CARD W/PARITY   (3   bytes) [0400                ]    
19382 ms <    +1 ms>:CODEC RX SNI READER          (2   bytes) [9320                ]    
19382 ms <    +0 ms>:CODEC RX SNI CARD W/PARITY   (6   bytes) [1122334444          ]    
19385 ms <    +3 ms>:CODEC RX SNI READER          (9   bytes) [93701122334444519c  ]    
19385 ms <    +0 ms>:CODEC RX SNI CARD W/PARITY   (4   bytes) [08b6dd              ]

geo-rg, 2019-09-02:
@sp4rrows Glad to see you figured it out yourself. :)