emsec/ChameleonMini #224: How to recover bootloader?

gtpy, 2019-07-30:
Hi,

how is the procedure to recover bootloader for RevG with external programmer?

JTAG? MKII? Other? pinout?

Has anybody an idea or a little guideline.

Thanks a lot.

fptrs, 2019-07-31:
Hi @gtpy ,
i think you need to flash the bootloader again. The bootloader firmware is [here](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Atmel%20DFU%20Bootloader/bootloader-atxmega128a4u-104-PA6-SPM.hex).

gtpy, 2019-08-02:
Ok thanks. But how do I flash it?

fptrs, 2019-08-02:
Hi @gtpy,
to flash the bootloader I use an AVR Dragon with the Atmel Studio Software. [Here](http://ww1.microchip.com/downloads/en/devicedoc/atmel-42723-avr-dragon_userguide.pdf) you can find the userguide.

gtpy, 2019-08-08:
Have you a picture how to connect it?

fptrs, 2019-08-09:
Take a look at the picture in section 4.3 of the AVR Dragon user guide. The other end of the cable is connected to the three pins at the bottom of the Chameleon and the read line points to the left. Additionally you have to connect the Chameleon via USB for power.

sp4rrows, 2019-08-11:
Hi @gtpy @fptrs maybe a beginner question:
Where to connect the chameleon RXD?
CLK and DATA I have found on the User Guide.

Would be nice if one of you could help me or expand the doxygen/wiki.

Peterthegreat, 2019-08-26:
Hi. Take a look at my older post. It's a bit of work, but I flashed the bootloader using a patched USBASP

https://github.com/emsec/ChameleonMini/issues/120#issuecomment-313001241

sp4rrows, 2019-08-30:
@Peterthegreat thanks, now I have a AVRispMKii.
But I am not sure how to exactly connect. 
Chameleon has 3 points (DATA, RXD, CLK).
My AVRisp has the following pinout:
[image]
MISO=DATA?
RST=CLK?
Thanks a lot.

Peterthegreat, 2019-08-31:
Yes, that should work. 
MISO=DATA
RST=CLK
I am not sure how, but you should change the AVRispMKii to PDI interface. Give it a try

sp4rrows, 2019-08-31:
PDI is possible.
What is RXD on the chameleon for? Anyone knows?

Peterthegreat, 2019-08-31:
I think my comment it's self explanatory. Just connect to the break-out pads. Do you know what breakout means? 

> Breakout boards are a common electrical components that take a bundled cable and “breaks out” 
> each conductor to a terminal that can easily accept a hook-up wire for distribution to another device.

The ChameleonMini has pads labeled CLK DATA etc. Take a closer look

The next comment (in that issue) by Endres is also helpful

gtpy, 2019-09-01:
@sp4rrows Try to take a look at the schematics. 
There you see, that on one side is DATA, RXD, CLK
And on the other you’ll find VCC, TXD, GND.
On the Board maybe a litte bit hidden on one side under some soldering.

gtpy, 2019-09-01:
So you should try a 2x3 Header to connect

closed 2019-09-02 by gtpy