emsec/ChameleonMini #226: Chameleon mini Rev G (MF Detection/Reader attack)

nico0481, 2019-08-14:
Hello,

Is the Chameleon mini Rev G able to do reader attack (MF Detection).
With which firmware?
I can't find a way to do this.

Thank you

Best regards

Nico

gtpy, 2019-08-14:
Not with standard firmware.
I had experimented with it a bit, hadn't tested it yet and hadn't finished it.
Need to check my local repos ;)

lavanoid, 2019-08-24:
I've attempted to bodge this feature into the current Rev G firmware and I'm clueless. I've asked the devs for the RevE Rebooted if they could assist with adding this feature to my fork of the Rev G firmware but they just tell me to go here. Would've been nice if they tried to help :/

https://github.com/lavanoid/ChameleonMini/commit/0ad00933b5aeca77cb0722e145d630c656dbe768

gtpy, 2019-08-25:
hi @lavanoid I tried the same time ago. found my local repo and pushed it now c8249d5 .

There I don't get problems with red LED. But I am not sure if its working.

Have no experience with detection function and are not sure what exactly is needed.
Maybe it is working, but I need some more tests or hints how to test.
Any Ideas?

I checked your code. Looks almost identical what I have done.

1. Set MF Classic & UID
2. Set Detection
3. Hold on reader
4. Use GUI and mfkey

lavanoid, 2019-08-25:
I just built the firmware with your commits and it seems to be working, however I have no idea what GUI to use to run mfkey. The latest build of ChameleonMini-rebootedGUI hides the mfkey option when detecting the Rev G.

I thought maybe something would be stored in the Chameleon device logs but I don't seem to have anything in there. 

This is the video I watched, to get an idea of how the MF_DETECT function works: https://www.youtube.com/watch?v=1VpXC3-eKhc

gtpy, 2019-08-25:
Try my GUI fort Rev G

https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI
It Support the mfkey.

-------- Original-Nachricht --------

lavanoid, 2019-08-25:
That seems to work! Requires a few retries and doesn't always find the keys (might just be because I'm using Mifare Classic Tool on Android and not an actual reader) but it does indeed work.

[image]
[image]
One bug with the GUI is that it doesn't let me change the UID of the tag, so I have to use another tool to do that.

gtpy, 2019-08-25:
UID changes worked. 
Enter UID --> Change UID.
If it doesn't work, give a short info, then I fix it.

How you check it with MCT? I never tried this :)

lavanoid, 2019-08-25:
```(doesn't work in detection mode - need to set for example mf_classic_1k)
If it doesn't work, give a short info, then I fix it.``` I'll give it a try when I'm booted into Windows again (only use Windows for games, so not signed into anything personal).

```How you check it with MCT? I never tried this :)```
I have a few key files from previous tags I have dumped. In MCT, I do this:

1.) Select "Read Tag"
2.) Select the key file I have, from a previous tag
3.) Select "Start mapping and read tag"

MCT will then attempt to authenticate with the Chameleon with the keys I told it to use, just like what a reader would do :)

gtpy, 2019-08-25:
Ok thanks. Need to use MCT more :)
Me too - only Linux user - Win only for games and some GUI stuff

lavanoid, 2019-08-25:
Greetings, my fellow Linux user 🐱

- Windows for games & password cracking with hashcat (pain controlling GPU overclocks and fan speeds on any Linux distro)
- Manjaro Linux for everything else except Netflix (cuz DRM)
- macOS for Netflix (faster than Windows, 1080p Netflix support on special titles, unix)

lavanoid, 2019-09-14:
Check out my repo for MF Detection support on the Rev G. It has had some changes made to it, for readers that check the card capacity.

Here is a pre-compiled version with the latest commits in my repo: https://github.com/lavanoid/ChameleonMini/releases/tag/BETA-RevG

ceres-c, 2019-11-01:
I think it might be time to close this issue @geo-rg @david-oswald

closed 2019-11-04 by fptrs

Walter-Bishop, 2020-01-30:
I tried to use gtpy's code in #229 but it seems their GUI is not available any more 😢: https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI 

@nico0481 do you (or anyone else) still have a copy of it and could upload it here?

I changed the iceman firmware so that it allows to use the mfkey32 attack but it just returns a CRC checksum error. I assume there is some difference in how the data is being transmitted in the Rev. E rebooted firmware and in #229 but I couldn't find any specific difference yet.