emsec/ChameleonMini #232: How to store data into an emulated card on the ChameleonMini Rev G

waleed-tariq, 2019-10-01:
I am doing a little project that requires me to test an rfid reader and one of the requirements is to have it be able to read the data in the card. I can get it to read the UID very easily, but I cannot figure out how to store data into it. I have tried to use the upload command and it seems to work but the data does not show. Perhaps I am doing it wrong. I would also like to store the data in a mifare classic 1k or a 4k card not the Ultralight. Thanks.

netvader, 2019-10-07:
If you want e.g. to emulate an Mifare Classic 4K 7B Card and you have an complete Dump, it should work. What exactly did you try?

waleed-tariq, 2019-10-09:
> If you want e.g. to emulate an Mifare Classic 4K 7B Card and you have an complete Dump, it should work. What exactly did you try?

So would a complete Dump be a .bin file. I found one on the RysCorp website that holds the binary card contents of a 4K card. When I do the upload it simply changes the UID of the card. Is it even possible to load data into a ChameleonMini? Also, can I write my own data files and upload them onto the Chameleon. I attempted to do so by creating a simple .txt file and wrote into it the data needed to pass a test. It was simply 8 letter 'o'. When I use XModem to upload it simply changes the UID to '6F' which is the hexadecimal equivalent to the ASCII key 'o'. Is this what is supposed to happen?

ceres-c, 2019-11-01:
If the upload is changing the UID of the card, then most likely it's also changing the data.
You can't use text files, you must know how data is stored inside the card (read Mifare datasheets) and then create a dump file with a hex editor such as 010Editor which has nice templates for the cards you're using.