emsec/ChameleonMini #234: Support for Legic Prime

nfc232, 2019-10-30:
Legic Prime is currently not supported. It would be nice, if it were possible to emulate Legic Cards.

david-oswald, 2019-10-31:
Yes, it certainly would - to my knowledge Legic Prime uses an own standard, which they actually tried to standardise as ISO14443 F, see https://events.ccc.de/congress/2009/Fahrplan/attachments/1506_legic-slides.pdf. You will need to write a corresponding codec to sniff or emulate, which is a relatively low-level coding task.

simplysoft, 2020-09-04:
Hi. We are fairly new to this project and have interest in Legic Prime emulation. Given that proxmark3 has working implementation of an emulator, is it doable to implement legic prime emulation for Chameleon? Or are there any limitations (e.g. in terms of hardware) that would make it more difficult to implement it with chameleon compared to proxmark3?
If there are no known road blocks, we might be able to invest some time into this.

david-oswald, 2020-09-07:
@simplysoft that would be really great, a long awaited improvement. Would be great if you could contribute something to the project in this regard.

I don't think there are fundamental issues, see here for the RF layer of Legic Prime:

[image]

The Chameleon HW should be able to pick up the 10% load mod and generate the reader modulation, however as usual the devil will be in the detail getting the timings right etc.

simplysoft, 2020-09-08:
@david-oswald is there any documentation or other resources that helps getting started with development for the chameleon HW? We did glance though the code and we are still a bit in the dark how to interact with the hardware.

ceres-c, 2020-09-08:
There was a workshop at the 36C3, I don't know whether it was recorded, but the slides are available [here](https://github.com/emsec/ChameleonMini/issues/244) 

Also, take a look at ISO14443-2A and ISO15693 codecs, they're well documented and you should be able to understand how data is de/modulated. I've improved ISO15 documentation, but the PR is not yet merged since I still have to fix a compilation error

simplysoft, 2020-09-09:
Thanks for the pointers so far. Any tips on development setup you use for testing / debugging or other tools you wouldn't not want to miss when writing code for Chameleon?

david-oswald, 2020-09-09:
you will definitely need an oscilloscope to check the waveforms on the demod pin etc I'd say. Doesn't have to be high-spec, something like a Rigol DS1054Z (or even cheaper ones) should do.

ceres-c, 2020-09-09:
I thought you might also want to have these 
Well you'll need

ceres-c, 2020-09-09:
I thought you might also want to have these Atmel application notes/manuals at hand while reading the code/developing your own

- 8045A-AVR-02/08
- 8071A-AVR-02/08
- 8331F–AVR–04/2013
- DS40002166A

The first two files are specific to interrupts/event system, 8331F is the manual for Xmega AU MCUs (handy to understand ports/interrupts registers) and the last one is the datasheet with all the stuff you'd expect to find in a datasheet and might need.

You'll need:

- An oscilloscope, as David sad. Being a student (≈ poor) I bought an OWON VDS1022I which proved to be sufficient for my needs. Of course a better scope would give better results and probably simplify debugging
- I found myself in need of a programmer to recover a bricked chameleon and opted for a [Atmel ICE](https://eu.mouser.com/ProductDetail/Microchip-Technology-Atmel/ATATMEL-ICE?qs=KLFHFgXTQiDAUrt43H15kQ%3D%3D) which double do as a [debugger](https://ceres-c.it/3debugging-ChameleonMini/). Again, being a student, I bought the [bare PCB](https://eu.mouser.com/ProductDetail/Microchip-Technology-Atmel/ATATMEL-ICE-PCBA?qs=KLFHFgXTQiA8s759Nx5NdQ%3D%3D) version of the debugger. Beware, it does not come with cables, so a you'll need a IDC cable (10 pins, 1,27mm pitch) to breakout your own connection board
- A reader/writer compatible with Legic cards, of course

street-grease-coder, 2021-06-26:
Why would a logic analyzer fall short here, compared to an oscilloscope? Asking because they're obviously orders of magnitudes cheaper (and because noob). Thanks for this great code repo.

timokasper, 2021-06-27:
a logic analyzer cannot display analogue signals; you need to see the real voltages at the pins of ChameleonMini in order to set up parameters such as threshold of the comparator, etc.

Lipovlan, 2026-01-05:
@simplysoft (or anybody, really) I hope code in my pull-request might serve as a good starting point for you.

I'll be working on some enhancements myself, I'd like to implement the LEGIC prime PRNG soon.