emsec/ChameleonMini #242: chameleon mini reve-2

nick500, 2019-12-06:
Can I upload firmware Chameleon Mini RDV2.0 ?

ceres-c, 2019-12-06:
Excuse me, what?
What is ChameleonMini RDV2.0? Are you referring to [ChameleonMini Rev.E Rebooted](https://github.com/iceman1001/ChameleonMini-rebooted)?

nick500, 2019-12-08:
> Простите, что?
> Что такое ChameleonMini RDV2.0? Вы имеете в виду [перезагрузку ChameleonMini Rev.E](https://github.com/iceman1001/ChameleonMini-rebooted) ?

Yes. My chameleon https://store.ryscc.com/blogs/news/34542273-unleash-the-chameleons            can i make him a sniffer or Chameleon Mini RDV2.0?

ceres-c, 2019-12-08:
If your chameleon uses the ATxMega32A4U MCU you should go [here](https://github.com/iceman1001/ChameleonMini-rebooted).
That repository has different features to this one

closed 2019-12-11 by fptrs