emsec/ChameleonMini #244: Chameleon Workshop at 36C3

fptrs, 2019-12-19:
[image]

Hi guys,
this year we are finally doing a workshop at 36C3 in Leipzig. If anyone has a wish regarding the content let us know. We will also update the workshops [page](https://events.ccc.de/congress/2019/wiki/index.php/Session:ChameleonMini_Workshop#_3a13f575e40c2951e1a26931802f0fb2) in the next days.

fptrs, 2019-12-27:
Here is a first draft of the [ChameleonMini programming guide](https://www.kasper-oswald.net:8042/f/f20d6875e8b0469188b2/?dl=1) presented at 36C3.

johndekroon, 2019-12-28:
Thnx for the awesome workshop. As a point of feedback I would suggest a more hands-on part, where people can practice with the Chameleon.

After the workshop I've a few questions:
- I put the card of my hotel room on the Chameleon, but it shows 203:TIMEOUT. I was able to interact with other cards. What can be the cause of this?
- Is it correct that we currently can only copy mifare ultralight cards?
- I put my Dutch ID card on the Chameleon, but the Identify command said "Unknown card type.". It was able to find the ATQA, UID and SAK. Is there any way to find more information about this card?
- At the presentation you mentioned that the top part can be cut of, did I understood that correctly?

Cheers!

Myszka, 2019-12-28:
Hey,
Do you have any stand/assembly on the 36C3? Where is it possible to find you and ask about ChameleonMini features?

ceres-c, 2019-12-28:
> Hey,
> Do you have any stand/assembly on the 36C3? Where is it possible to find you and ask about ChameleonMini features?
I am not one of the original devs but I've been writing quite some code for the Chameleon. You can find me strolling around but we can meet @ Italian Hackers Embassy if some help is needed.

ceres-c, 2019-12-28:
> I put the card of my hotel room on the Chameleon, but it shows 203:TIMEOUT. I was able to interact with other cards. What can be the cause of this?

Were you trying to identify or clone the card? Identification should be working if the card is ISO14443A. It could be a ISO14443B or ISO15693 card, or maybe even a low frequency card (125/134 kHz), while cloning only works with Mifare Ultralight.
> Is it correct that we currently can only copy mifare ultralight cards?

If I'm not mistaken: yes. For mifare Classic cards you need the keys which can't be obtained with the Chameleon.

> I put my Dutch ID card on the Chameleon, but the Identify command said "Unknown card type.". It was able to find the ATQA, UID and SAK. Is there any way to find more information about this card?

You can use an android phone or a Proxmark, but all in all you just need to search those bytes you found (ATQA and SAK) online and you should be able to manually identify the tag you're dealing with.

> At the presentation you mentioned that the top part can be cut of, did I understood that correctly?

Yes, the "detached" part can be cut off, but that'd prevent your chameleon from working inside long card readers which have a switch on the back as the switch will not be clicked

fptrs, 2019-12-28:
Hi everyone,
I just talked to @ceres-c. We will do another sale of the Chameleons at the Italian Hackers Embassy tomorrow between 3 and 5 pm. We'll also be around to do some Q&A

ceres-c, 2019-12-30:
I know it's a close call, but me and @fptrs are holding a talk about breaking a NFC Toy with the ChameleonMini at the Wikipaga WG stage in 30 minutes. It's starting at 13:15.

fptrs, 2019-12-31:
For those of you who did not attend the very spontaneous talk, here is the [link ](https://streaming.media.ccc.de/36c3/relive/608)to the recording. We started at minute 20.