emsec/ChameleonMini #248: Windows drivers fail to start chameleon revG

nov3mb3r, 2020-01-01:
I just downloaded the drivers from the project and installed them in Device Manger. Installation proceeded without errors, but upon reboot, this error appears.
```
This device cannot start. (Code 10)
The specified request is not a valid operation for the target device.
```
TeraTerm now does not recognize the serial port to let me pass commands

johndekroon, 2020-01-02:
No Windows here, so I can't help. However, what you can do is install virtual box and use a VM with Linux. This should work fine.

fptrs, 2020-01-03:
Hi @nov3mb3r,
I installed the Windows driver as well, never had such problems. Did you try to reinstall it? You might also want to read [this](https://www.lifewire.com/how-to-fix-code-10-errors-2623181).