emsec/ChameleonMini #256: UID can be changed with NFC directly

whywilson, 2020-02-03:
After emulating m1 card, block 0 of all 8 slots can be changed with Android NFC hardware.
How to make it become ordinary m1 card?
Please help.

fptrs, 2020-02-07:
Hi @whywilson,
if you want to emulate a Mifare Classik 1k card set ```config=MF_CLASSIK_1K```. Next set ```uid=<4 Byte UID>``` or upload a card dump in the .mfd format.

whywilson, 2020-02-12:
Hi @fptrs ,
Thank you for your reply.
There are no issues for me with emulating a Mifare Classic 1k card.
The problem is the block 0 of sector 0 can be modified if I write with another dump. Just like CUID card.

fptrs, 2020-02-12:
Hi @whywilson,
by default the firmware is compiled with ```SUPPORT_MF_CLASSIC_MAGIC_MODE```. This enables the command set for mifare classic clones which support overwriting the first sector. Try to compile your firmware without this flag.

Akisame-AI, 2020-03-12:
Does the revG store the modifications you write directly to its virtual card?
I want to emulate a card (mifare classic 4k) that receives a code to sector 1 that is used again when reading it again.

fptrs, 2020-03-17:
Yes modifications are stored on the chameleon. You could prevent this by setting it to read only mode.

closed 2021-03-31 by david-oswald