emsec/ChameleonMini #257: Get dump from ISO15693 and emulate.

AngelSotomayor, 2020-02-20:
Hello,

I was watching the video posted in the #244  issue and I saw that @ceres-c and @fptrs did break a NFC Toy that uses the ISO15693, and that is exactly that I need to do, but as a beginner using the chameleon I'm a bit lost with the proccess.
I'm capable of sniffing the comunication between a reader and the tag, but I'm kinda lost with the next steps.
How can I read the data inside the tag with the chameleon? How can I extract a dump of the content of the ISO15693 tag? Once I gat this dump I will only need to upload it to one slot of the Chameleon and I will be emulating the tag right?

Many thanks for any help you can provide!

I will continue looking through the guides, this github and toying with the chameleon mini in case that I found the answers myself.

fptrs, 2020-02-21:
Hi @AngelSotomayor,

the FW does not feature an ISO15693 reader yet. We used another reader to extract the content of the tag. But we did not need this content to emulate the toy, we just used the UID recorded with the sniffer. Once you have the UID and/or the content you can upload it to the chameleon and emulate a ISO15693 tag. Do you know the type/manufacturer of your tag? Can you upload a log of the sniffed communication?

AngelSotomayor, 2020-02-24:
Thank you so much for your quick answer.

The tag is a custom Texas Instrument based on the RF430FRL152H.
I can post the log in a while.
Regards

AngelSotomayor, 2020-02-25:
Sorry for being late, yesterday I didn't have a mobile with NFC to scan the tag.
This is the lof of the sniffed data. What I'm doing is read the entire memory with tha app NFC TagInfo.
Here is the result:

[24414] CODEC RX DATA: 26 01 00 F6 0A
[24444] CODEC RX DATA: 22 2B EE DC 6C 00 00 A0 07 E0 45 B7
[24467] CODEC RX DATA: 22 20 EE DC 6C 00 00 A0 07 E0 00 26 F8
[24515] CODEC RX DATA: 22 2B EE DC 6C 00 00 A0 07 E0 45 B7
[24532] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 00 23 35
[24546] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 01 AA 24
[24563] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 02 31 16
[24580] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 03 B8 07
[24596] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 04 07 73
[24611] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 05 8E 62
[24627] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 06 15 50
[24647] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 07 9C 41
[24667] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 08 6B B9
[24679] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 09 E2 A8
[24692] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0A 79 9A
[24710] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0B F0 8B
[24725] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0C 4F FF
[24740] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0D C6 EE
[24754] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0E 5D DC
[24767] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 0F D4 CD
[24779] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 10 A2 25
[24801] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 11 2B 34
[24813] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 12 B0 06
[24832] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 13 39 17
[24851] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 14 86 63
[24864] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 15 0F 72
[24877] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 16 94 40
[24892] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 17 1D 51
[24912] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 18 EA A9
[24923] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 19 63 B8
[24936] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1A F8 8A
[24948] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1B 71 9B
[24967] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1C CE EF
[24979] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1D 47 FE
[24990] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1E DC CC
[25011] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 1F 55 DD
[25027] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 20 21 14
[25038] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 21 A8 05
[25050] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 22 33 37
[25062] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 23 BA 26
[25073] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 24 05 52
[25085] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 25 8C 43
[25096] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 26 17 71
[25109] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 27 9E 60
[25120] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 28 69 98
[25136] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 29 E0 89
[25277] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2A 7B BB
[25299] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2B F2 AA
[25328] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2C 4D DE
[25340] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2D C4 CF
[25351] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2E 5F FD
[25364] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 2F D6 EC
[25375] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 30 A0 04
[25387] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 31 29 15
[25398] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 32 B2 27
[25412] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 33 3B 36
[25423] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 34 84 42
[25435] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 35 0D 53
[25446] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 36 96 61
[25504] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 37 1F 70
[25518] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 38 E8 88
[25529] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 39 61 99
[25544] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 3A FA AB
[25559] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 3B 73 BA
[25570] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 3C CC CE
[25581] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 3D 45 DF
[... 172 lines cut ...]
[27773] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 EA 77 7D
[27785] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 EB FE 6C
[27796] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 EC 41 18
[27808] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 ED C8 09
[27820] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 EE 53 3B
[27832] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 EF DA 2A
[27844] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 F0 AC C2
[27856] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 F1 25 D3
[27868] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 F2 BE E1
[27879] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 F3 37 F0
[27891] CODEC RX DATA: 62 20 EE DC 6C 00 00 A0 07 E0 F4 88 84
[28047] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28205] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28365] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28524] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28682] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28841] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[28999] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[29158] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[29317] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[29475] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[29635] CODEC RX DATA: 26 01 40 EE DC 6C 00 00 A0 07 E0 5F AC
[29821] CODEC RX DATA: 26 01 00 F6 0A

Regards

fptrs, 2020-02-26:
Hi @AngelSotomayor,
the app reads each block of the tag with ```6220<UID><Block number><CRC>```. You can use the information shown in the app to emulate this tag with the EM4233 config as a starting point.
1. Create a binary dump containing the data you read with the app.
2. Set the config.
3. Upload the dump.
4. Set the UID.

One remaining difference between the emulation of the EM4233 config and your tag is the answer to the ```GetSystemInformationCommand``` (```22 2B EE DC 6C 00 00 A0 07 E0 45 B7```). This command expects for instance the number of content blocks, the block width and the DSFID. These values are also shown in the NFC TagInfo App.
A first step is to adjust these values in the EM4233 config. Then you compare the emulation of the chameleon and your tag. Finally you could create a new configuration for your tag that uses these explicit values.
Keep me updated on your progress and have fun.

stappjno, 2020-07-16:
Have you done any progress here? I found this issue and think this is similar to my questions:
https://github.com/emsec/ChameleonMini/issues/265

alvarotorijano, 2023-01-05:
hi @AngelSotomayor did you achieved to emulate the 15693 tag?