emsec/ChameleonMini #260: BOM correction (for selfmade hw)

robots, 2020-04-10:
Hi,
for all of you guys who made Chameleon Mini on their own here is little advice:

Capacitors C29 and C30 should be 47pF.

This tunes the mismatch between transmitter and antenna. It increases reader distance from -1 to about 2-3cm. RSSI without tag changes from 500-600 to ~2500mV.

More about this here: https://my.eng.utah.edu/~mlewis/ref/NFC/AN1445.pdf

timokasper, 2020-04-10:
Hi robots, Thanks for the advice, but I cannot recommend to follow your advice. The RSSI without tag, with exactly the components of the BOM, should be more than 10V - all of the ChameleonMini RevG sold in the webshop of Kasper&Oswald have to pass this test! It seems that you have made some other mistake during manufacture. I have spent about a week for fine-tuning the antenna matching both for emulation and reader operation of Rev.G, before we had several thousand devices produced after the kickstarter project...  and changing the value of a single (or even 2) capacitors affected all capabilities (sniffing, reading, emulating, ...) it was a tidious try-and-improve approach and I think we have found a good compromise for the tuning....     anyway, 2500mV is definitely not enough, there is s.th. wring with your device.

robots, 2020-04-10:
Well, its been 2years since i made the board. I dont remember what parts were ordered, but it could be problem on my side :-) I will replace all RF frontend parts just to be sure.

closed 2020-04-10 by robots

libin-ka, 2020-05-04:
What is the capacitance value of C29 C30

robots, 2020-05-04:
these capacitors are not populated in BOM

libin-ka, 2020-05-04:
[image]

libin-ka, 2020-05-04:
> 这些电容器未填充在BOM中
See pictures c29 and c30 in the wiki
https://github.com/emsec/ChameleonMini/wiki