emsec/ChameleonMini #263: ISO14443A_SNIFF

madalinaStreche, 2020-05-27:
I try to record a communication between 2 devices, but from my logs it seems that Chameleon manages to sniff only in one direction: reader -> card. What about the transmitted data from the card to the reader? Are there any updates on this?

fptrs, 2020-06-03:
Hi @madalinaStreche,
did you already take a look at #220?

madalinamarin, 2020-08-13:
> Hi @madalinaStreche,
> did you already take a look at #220?

Hi! I managed to update the Chameleon and now it works properly. Thanks!

madalinamarin, 2020-08-13:
Hi! I managed to update the Chameleon and now I can see the logs from both sides (card/reader). The other issue that I have right now is the fact that some messages are incomplete, altered or modified. I tried to record an EMV communication between a smartwatch and a Raspberry PI that acts like a POS. I sniff with the Chameleon the same communication for 20 times, each communication has in total 14 messages(7 from the Raspberry and 7 responses from the smartwatch) and create a statistics which suggests the frequency of altered/missing/incomplete messages. Do you have any idea why this is happening?
[image]

madalinamarin, 2020-08-13:
Hi! I managed to update the Chameleon and now I can see the logs from both sides (card/reader). The other issue that I have right now is the fact that some messages are incomplete, altered or modified. I tried to record an EMV communication between a smartwatch and a Raspberry PI that acts like a POS. I sniffed with the Chameleon the same communication for 20 times, each communication has in total 14 messages(7 from the Raspberry and 7 responses from the smartwatch) and I created a statistics which suggests the frequency of altered/missing/incomplete messages. Most messages that have problems are responses from the smartwatch.
Do you have any idea why this is happening?
[image]

madalinaStreche, 2020-08-13:
Hi! I managed to update the Chameleon and now I can see the logs from both sides (card/reader). The other issue that I have right now is the fact that some messages are incomplete, altered or modified. I tried to record an EMV communication between a smartwatch and a Raspberry PI that acts like a POS. I sniffed with the Chameleon the same communication for 20 times, each communication has in total 14 messages(7 from the Raspberry and 7 responses from the smartwatch) and I created a statistics which suggests the frequency of altered/missing/incomplete messages. Most messages that have problems are responses from the smartwatch.
Do you have any idea why this is happening?
[image]

ceres-c, 2020-08-13:
It might be due to autocalibration not being perfect, so quality of sniff depends on the physical position of the chameleon in the NFC field

david-oswald, 2020-08-14:
@madalinaStreche Sniffing in the direction from reader to card should normally work without problems. The opposite direction is much harder due to the weaker load modulation. So it might depend on setup, calibration etc as the others pointed out.

madalinamarin, 2020-08-14:
The autocalibraton mode is available only for reading mode? It will help if I will run the autocalibration in reader mode first and then change to sniffing? 
Also, I just tried to sniff a real communication between smartwatch and POS during a payment and the statistics are really bad, 95% are altered/missing messages. What is different in this case? I kept the Chameleon right between the smartwatch and POS.

ceres-c, 2020-08-14:
https://github.com/emsec/ChameleonMini/issues/220#issuecomment-494812896