emsec/ChameleonMini #265: Cant find script mentioned in presentation

stappjno, 2020-07-08:
Where can I find the script to clone one of the figurines?

https://media.ccc.de/v/36c3-108-hacking-an-nfc-toy-with-the-chameleonmini#t=0

ceres-c, 2020-07-08:
It's on my gist account
https://gist.github.com/ceres-c/cbf437f9af9e946c96550fd7e1e77cef

stappjno, 2020-07-14:
Thank you! Is there a tutorial how to read and emulate the ids of my figurines? I am new to chameleonmini.

fptrs, 2020-07-15:
Hi @stappjno,
here is the [firmware](https://github.com/ceres-c/ChameleonMini/tree/ICODE-SLI/Firmware/Chameleon-Mini/Latest) you need to use on the chameleon. You need to flash the chameleon and simply follow the steps in the video. @ceres-c once you find the time you could finally create the pull request and I will merge it 😄

ceres-c, 2020-07-15:
Hi @fptrs, long time no see!
I haven't yet made a PR since I did not remember wether the code could be considered stable/final or it needed some improvements. You probably remember better than me, since you were the author of that part of the code :)

stappjno, 2020-07-15:
I tried to run the script (win10 with ubuntu 10.04 as subsystem), installed all requirements (Serial Port is available on COM3) but I get this error
```
Traceback (most recent call last):
  File "CR95HF_ICODE_psw_dump.py", line 76, in <module>
    h.open(0x0483, 0xd0d0)
  File "hid.pyx", line 66, in hid.device.open
OSError: open failed
```

ceres-c, 2020-07-15:
Sorry, why are you using WSL when you could use python straight on your windows machine? Also, are you sure the port is accessible to the WSL? This is not the case by default, AFAIK

stappjno, 2020-07-16:
I've tried this but hid does not run very well on windows (or I can't get it to work with the dll). I think I have access to the interface (i can access the serial console with ubuntu) via ttyS3

stappjno, 2020-07-16:
Ok I have rewatched the conference video and saw that you are using two types of hardware. I want to achieve two things:
 - Read the ids of tonies
 - Emulate a tonie with the id i've got
I have just the chameleonmini and I thought that both steps could be done with the chameleonmini revG. Is that wrong? 

The script is to obtain the uid of a tag and seemed to me to work with the chameleonmini. Maybe I understood something wrong here? Or is it possible to obtain the id while sniffing the communication between the box and the figurine?

I dont find some docs how I can emulate a figurine with a given ID and as I understand the regular way would be to write the id to slot 1 and set the mode to the type which corresponds to the tag achitecture. The only available mode I can see is ISO15693_SNIFF which probably wont be able to emulate anything. I know these are much questions and this issue isn't the right place for them but maybe you can help me out :D 

Thanks in advance

ceres-c, 2020-07-16:
Hi,
Yes, that's correct, we were using both a Chameleon and a ST M24LR Discovery board. The former to emulate the tag with the appropriate ID (and to obtain the password) and the latter to read the tag inside the tonie.
ISO15693_SNIFF is not able to sniff bidirectional communication and there is no public implementation of a ISO15693 reader mode for the Chameleon. Your best bet is to buy the ST device or port the python script to whatever other NFC reader with iso15 support you own.

Once you have a dump for your tag, to emulate the figurine you should use the firmware @fptrs linked (the ICODE branch in my repo) and use the ICODE config for the Chameleon.

stappjno, 2020-07-16:
Ok if I understand you right it's just possible to emulate a figurine with the chameleonmini. But that would statisfy one of my needs ;) Lets focus on this. I have flashed the mentioned firmware and I can activate the config
```
CONFIG=ICODE_SLI
```

after that I set the ID of a tonie tag

```
UID=223a4615200103e0
```

When I now put the chameleonmini on the box nothing happens. Do I miss something?

ceres-c, 2020-07-16:
Yes, you understood correctly.
If I'm not mistaken you'll also need the content of the tag, the UID won't be enough. To get the content you'll need to read the tag.

Sorry if I can't give you more precise information but I don't own a Tonie and I haven't ever seen one after I left the Congress on 30 December 2019. Also, I didn't get much sleep those days, so everything's a bit blurry :)

stappjno, 2020-07-16:
Thank you for your great (and very fast!) support :) I'll oder something to read the tags. One last thing: Do you maybe have the payload of one of the tonies you used at the presentation? That would be great to start experimenting. If you dont want to post it publicly you could send me an email at claudio.goetz@gmail.com

ceres-c, 2020-07-16:
Have a look here https://github.com/toniebox-reverse-engineering/teddy
I don't think I have any dump, but I wouldn't share them anyhow since the toy we dumped wasn't mine. Sorry, you'll have to wait.

PS eBay is a fine source for that demo board, I bought mine there from Germany at a competitive price

fptrs, 2020-07-20:
Hi @stappjno,
Once a Tonie is downloaded to your box, you do not need the tags content anymore. The UID is sufficient. You can obtain the UID by sniffing the reader to tag communication, since the Tonie box uses the ISO15 adress mode. Then the ICODE config should work. Can you post the log of your chameleon emulating the UID? Maybe you put in the UID in the wrong byte order.

netvader, 2020-09-16:
After flashing the latest commits on Aug 27, 2020 [f62c8fd](https://github.com/ceres-c/ChameleonMini/tree/f62c8fdd73ecaaee5e8f59d3d0cacdc4227994ed) the ICODE_SLI implementation with UID simulation for my Toniebox doesn't work for me anymore. The [firmware](https://github.com/ceres-c/ChameleonMini/tree/ICODE-SLI/Firmware/Chameleon-Mini/Latest) version linked above does not work either, in this firmware is no ICODE config mode. I thought there was another ICODE_SLI code base from end of December. Unfortunately I can no longer find these code base which worked for me. I only find commit 22023b7 that doesn't have that ICODE implementation. Please correct me if I am absolutely wrong right now.

Unfortunately I don't know what to do next, I am a little surprised that the ICODE only UID simulation with ChameleonMini worked before flashing the latest commits. I tried different owned and already known tonies. I also tried it with and without 32 Byte long memory content. Maybe these informations helps, this is a current 

> comparison. The UID and SYSINFO is the same every time and deliberately hidden, the only difference I find out is the memory layout.

**original known tonie**
`
[usb] pm3 --> hf 15 info u

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]       TYPE: NXP(Philips); IC SL2 ICS50/ICS51(SLI-L) ICS5002/ICS5102(SLIX-L)
[+]        UID: E0 04 03 xx xx xx xx xx
[+]    SYSINFO: xx xx xx xx xx xx xx xx xx xx xx xx xx xx
[+]      - DSFID supported        [0x00]
[+]      - AFI   supported        [0x00]
[+]      - IC reference supported [0x03]
[+]      - Tag provides info on memory layout (vendor dependent)
[+]            4 (or 3) bytes/blocks x 8 blocks
`
**with ChameleonMini**
`
[usb] pm3 --> hf 15 info u

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]       TYPE: NXP(Philips); IC SL2 ICS50/ICS51(SLI-L) ICS5002/ICS5102(SLIX-L)
[+]        UID: E0 04 03 xx xx xx xx xx
[+]    SYSINFO: xx xx xx xx xx xx xx xx xx xx xx xx xx xx 
[+]      - DSFID supported        [0x00]
[+]      - AFI   supported        [0x00]
[+]      - IC reference supported [0x03]
[+]      - Tag provides info on memory layout (vendor dependent)
[+]            4 (or 3) bytes/blocks x 16 blocks
`

timokasper, 2020-09-16:
I confirm that recently the UID-only emulation doesn't work with my TonyBox, seems like there has been a firmware upgrade of the box? Probably we have to move forward to "full emulation" of the tags.

netvader, 2020-09-16:
Thanks for the feedback and to the developers so far. 
According to my chamlog results, I suspect that a possibly new box firmware may now also check "random number" (and maybe privacy mode) answer and this function does not seem to be supported by the Chameleon ICODE_SLI implementation yet, but i'm not an expert, so could be wrong. so at some point a full emulation of the tags would be great. 😄

ceres-c, 2020-09-19:
When I began updating the ICODE fork I was aiming for (mostly) complete emulation, but then I got drifted away by other issues and interrupted development.
Being too confident, I merged all the old commits and my new changes into this single commit https://github.com/ceres-c/ChameleonMini/commit/f62c8fdd73ecaaee5e8f59d3d0cacdc4227994ed but, as you found out, I implemented something wrong and we hit a regression.
It might be due to the wrong memory layout, as you pointed out. Can you confirm original tonies (I don't own any, so I'm working blind) have 8 blocks? According to the [NXP datasheet](https://www.nxp.com/docs/en/data-sheet/SL136430.pdf) they should have 16.

netvader, 2020-09-19:
@ceres-c thanks for your response and your work, I appreciate that! Yes you are right, the should have 16 blocks, but i can also confirm that all my own testet tonies have 8 blocks, at least that's what proxmark says. If you could provide sometime an 8 block special implementation, I would be happy to test it with my Chameleon.

ceres-c, 2020-09-19:
Let's try this out @netvader 
https://github.com/ceres-c/ChameleonMini/tree/ICODE-SLI

netvader, 2020-09-22:
@ceres-c Thanks for the quick help. I just tried it quickly. The block size fits now, but unfortunately it still doesn't work. But I'll try again this week when I have more time and contribute a few LOGs.

ceres-c, 2020-09-22:
Thanks
If we won't be able to fix this issue with logs I'll end up buying a tonie figurine. It's not like I'll do much with it once emulation is done, but I want to fix this mess I've made.

If you can, send me the log unedited to my email address (you can find it on my gh profile)

timokasper, 2020-09-23:
I have Tonie box + Tonies here and can help, also Fabi can help, he is currently busy with holidays et cetera :D :D

netvader, 2020-09-23:
@ceres-c i send you some logs directly via keybase, i hope that works also ... ;)

Ramblurr, 2022-02-11:
Hi folks! My use case is to use old tony figures in my self-built gadget similar to a toniebox. I only want to use the tonies RFID to trigger my own audio, so I just need it to function like a normal/open RFID tag.

Is the following possible?

"unlock" the tonie using the Chameleon, then read the tag at will using any normal non iso15693 RFID reader that operates at  13,56 MHz (such as the RC522 that implements iso14443)?

Or even after being unlocked, can the tag only be read by iso15693 implementing readers?

stappjno, 2022-02-11:
Hi @Ramblurr 
We have a nice community of toniebox-modders here: https://t.me/toniebox_reverse_engineering
I think the chameleon is not able to unlock the figurines (and they will be unlocked each time you put them back on the original box). A proxmark3 can unlock them and read all necessary data (with TeddyBench). You can also emulate tonies with it.

ceres-c, 2022-02-11:
You can't unlock the tag with the chameleon since the chameleon is not an iso15693 reader
You're still going to need an ISO15693 reader once the tag is unlocked: iso14443 and iso15693 differ greatly on a air interface level, thus the two are absolutely incompatible