emsec/ChameleonMini #266: Unable to recognize emulated tags using nfc-list from libnfc

maxieds, 2020-07-09:
I am trying to get the ``nfc-list`` utility from ``libnfc`` to recognize tags emulated by the Chameleon Mini. So far I am having no luck. Is there a known reason why this doesn't work? 

I am using an ACR122U reader up and running with ``libnfc``. When I run the following command, none of the emulated tag configurations (MF_CLASSIC, MF_ULTRALIGHT, etc.) end up getting recognized:
```bash
sudo LIBNFC_LOG_LEVEL=3 nfc-list -v
```

david-oswald, 2020-07-09:
We have been using `nfc-list` / `libnfc` for years to demonstrate and develop the Chameleon Mini, so it should (tm) work. Have you tried adjusting the position reader vs Chameleon?

maxieds, 2020-07-09:
@david-oswald 
Here is the result of running ``nfc-list`` on the latest firmware running a MFC configuration:
```bash
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc libnfc-1.8.0-30-g66d3560
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 041 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:041"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00 
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00 
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00 
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02 
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00 
debug	libnfc.chip.pn53x	SetParameters
debug	libnfc.driver.acr122_usb	TX: 6f 08 00 00 00 00 00 00 00 00 ff 00 00 00 03 d4 12 14 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 13 90 00 
debug	libnfc.general	"ACS / ACR122U PICC Interface" (acr122_usb:001:041) has been claimed.
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD False
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 11 00 00 00 00 00 00 00 00 ff 00 00 00 0c d4 06 63 02 63 03 63 0d 63 38 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 09 00 00 00 00 00 00 81 00 d5 07 83 83 00 01 07 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_BitFraming (Adjustments for bit oriented frames)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0a 00 00 00 00 00 00 00 00 ff 00 00 00 05 d4 08 63 3d 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.general	set_property_bool NP_ACTIVATE_FIELD True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 32 01 01 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.general	set_property_bool NP_AUTO_ISO14443_4 True
debug	libnfc.general	set_property_bool NP_FORCE_ISO14443_A True
debug	libnfc.general	set_property_bool NP_FORCE_SPEED_106 True
debug	libnfc.general	set_property_bool NP_ACCEPT_INVALID_FRAMES False
debug	libnfc.general	set_property_bool NP_ACCEPT_MULTIPLE_FRAMES False
NFC device: ACS / ACR122U PICC Interface opened
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT False
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 13 00 00 00 00 00 00 00 00 ff 00 00 00 0e d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 0a 00 00 00 00 00 00 81 00 d5 07 83 83 40 00 10 00 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 08 63 02 80 63 03 80 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 00 01 02 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 4b 00 90 00 
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
[... 72 lines cut ...]
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 4b 00 90 00 
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
1 ISO14443A passive target(s) found:
[... cut ...] get:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): 37  4d  a0  3d  
      SAK (SEL_RES): 08  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Classic 1K
* MIFARE Plus (4 Byte UID or 4 Byte RID) 2K, Security level 1
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:
```
I tried adjusting the reader distances.

david-oswald, 2020-07-09:
Can you take a log on the Chameleon to see if something arrives and/or where the reader aborts?

maxieds, 2020-07-09:
@david-oswald 
This is what I get out of the *Chameleon Mini Live Debugger* with ``LOG_MODE=LIVE`` and a MFC-4K configuration:
```bash
NFC device: ACS / ACR122U PICC Interface opened
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT False
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 13 00 00 00 00 00 00 00 00 ff 00 00 00 0e d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d 
debug	libnfc.driver.acr122_usb	RX: 80 0a 00 00 00 00 00 00 81 00 d5 07 83 83 40 00 10 00 90 00 
debug	libnfc.chip.pn53x	PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 08 63 02 80 63 03 80 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 09 90 00 
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 00 01 02 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
debug	libnfc.chip.pn53x	InListPassiveTarget
debug	libnfc.chip.pn53x	Timeout value: 300
debug	libnfc.driver.acr122_usb	TX: 6f 09 00 00 00 00 00 00 00 00 ff 00 00 00 04 d4 4a 01 00 
debug	libnfc.driver.acr122_usb	RX: 80 05 00 00 00 00 00 00 81 00 d5 4b 00 90 00 
debug	libnfc.general	set_property_bool NP_INFINITE_SELECT True
debug	libnfc.chip.pn53x	RFConfiguration
debug	libnfc.driver.acr122_usb	TX: 6f 0b 00 00 00 00 00 00 00 00 ff 00 00 00 06 d4 32 05 ff ff ff 
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00 
0 ISO14443A passive target(s) found.
```
And debugging / logging output:

[image][image]

geo-rg, 2020-07-09:
@maxieds Have you tried emulating another UID (e.g., not containing ff-bytes only).

Also, either the live debugger tool or the Chameleon log seems to have a problem, since the first codec_rx consists of 7 bytes even though it should be only 1 (0x26 = REQA). You can even see the following log entry starting with 0x41 being the log type.

maxieds, 2020-07-09:
@geo-rg 
Yes, I tried it on a different UID. The same lack of recognition happens. Also, there is a slight bug in the way the CMLD application is parsing the logs.

One thing I'm wondering looking at the ISO1443_SNIFF output when reading my known MFC tag is whether the ATQA value should have some kind of a parity bit transferred with it? In the output below, it's not just transferring ``0400`` (ATQA=0004) it's actually sending ``04 00 02``. This seems to make ``nfc-list`` pick up the tag. 

Any suggestions?

[image]

maxieds, 2020-07-10:
Can anyone verify that the current firmware is working with a particular tag reader under ``libnfc``? I need to rule out screwy behavior from my new ACR122U reader. It has been known to have recent oddities with different packaging. 

Thanks.

fptrs, 2020-07-15:
Hi @maxieds,
I testet the current firmware with `nfc-list` on Ubuntu using an SCL3711 reader and on Windows using the ACR122U. Both readers pick up the tag.

maxieds, 2020-07-31:
@fptrs 
I'm still having some timing issues, but I think I pinned down the problem. The Chameleon was (is) getting stuck in the anti collision loop due to timeouts. There was one feature / fix I added to my CMLD app that I have been using to log the output. Basically, while we're doing time sensitive things responding to the anti collision commands, the app is repeatedly refreshing the data in the toolbar, which invokes a number of slow USB data transfers for a while via the Chameleon command line. 

I'm now getting the following output for a MFC tag emulated by the Chameleon with all of that updating disabled in the logger app:
```bash
$ sudo nfc-anticol 
NFC reader: SCM Micro / SCL3711-NFC&RW opened

Sent bits:     26 (7 bits)
Received bits: 04  00  
Sent bits:     93  20  
Received bits: ff  ff  ff  ff  00  
Sent bits:     93  70  ff  ff  ff  ff  00  27  d0  
WARNING: Cascade bit set but CT != 0x88!
Sent bits:     95  20  
Sent bits:     95  70  ff  ff  ff  ff  00  ea  88  
WARNING: Cascade bit set but CT != 0x88!
Sent bits:     97  20  
Sent bits:     97  70  ff  ff  ff  ff  00  51  bf  
Sent bits:     e0  50  bc  a5  
Sent bits:     50  00  57  cd  

Found tag with
 UID: ffffffffffffffffffff
ATQA: 0004
 SAK: ff
```
You all can close the issue if you like now that I have verified things as working on my end.

closed 2020-08-07 by fptrs