emsec/ChameleonMini #270: Chameleon not getting into Bootloader in Windows 10

nov3mb3r, 2020-08-12:
After following all the steps, installing the Chameleon driver, teraterm and dfu programmer, i have a problem entering the bootloader. I press the RButton, upon connecting my USB, but chameleon lights never change. Is this supposed to be like that or could my chameleon be faulty?? I used 2 different computers and installed drivers, but i can't get into bootloader.
When I reboot my computers, The Chameleon driver returns a code 10 error
[image]

madalinamarin, 2020-08-13:
I had the same problem, I solved it using macOS.

david-oswald, 2020-08-14:
Dear @nov3mb3r, can you check in the dev mngr that the atmel drivers are installed and that the chameleon is really in the bootloader (i.e. shows up as Atmel DFU or so)?

nov3mb3r, 2020-08-14:
This is my Device Manager:
[image]
I would expect that the bootloader is the AT32UC3A. Is that correct? In my understanding, drivers are installed but somehow i can't get to boot

timokasper, 2020-08-14:
are you using an original ChameleonMini RevG? Then this should be an ATXMEGA128A4U. For other devices please ask the manufacturer.

nov3mb3r, 2020-08-14:
@timokasper As I said in the first issue, I use an original ChameleonMini RevG. I bought it from your workshop in 36c3

timokasper, 2020-08-14:
just to be sure: the chameleon is switched OFF, then you hold the Rbutton, then you switch ON again ; as a result no LED lights up (indicating the bootloader is active) and then you plug in the USB, correct?

nov3mb3r, 2020-08-14:
Correct!

timokasper, 2020-08-14:
[image]

I did the same it looks like this. Windows 10 64 bit.

so for some unknown reason you have a different driver. I had installed a while ago the WinAVR package is that maybe the cause?

nov3mb3r, 2020-08-14:
No I have not installed WinAVR or other packages. I retested in a laptop where I have practically nothing, and I have the same problem

ceres-c, 2020-08-14:
This might be a bit too barbaric, but I'd give a try with [Zadig](https://zadig.akeo.ie/). Try changing the default driver for the Chameleon to libusb, but PLEASE pay attention to not fu*k up any other device

nov3mb3r, 2020-08-14:
@ceres-c I would appreciate some instructions since I am a total newbie in hardware

ceres-c, 2020-08-14:
It should be pretty straightforward, you have to select the device you're 100% sure to be the chameleon from the dropdown menu and then pick "libusb" in the selector to the right of the green arrow. Then use the big button "install libusb" (or something like that). The chameleon should then be automatically soft-unplugged from your PC and the device list from device manager will be updated. There you should be able to see something like the output @timokasper has "libusb-win32".
Just be sure to select the right device to force driver update on (the name should be the same as in Device Manager)

nov3mb3r, 2020-08-14:
@ceres-c Interestingly enough, there are no devices to choose apart from my keyboard and mouse! I am starting to suspect that my chameleon might be faulty

ceres-c, 2020-08-14:
Check the "List all devices" checkbox in options menu

nov3mb3r, 2020-08-14:
I already did that before posting :)

ceres-c, 2020-08-14:
That's quite strange indeed.
It seems a driver issue to me, since an Atmel device is correctly recognized by windows. Could you try with a linux live iso? I can't help you further since USB issues with this kind of devices were one of the reasons I switched from Win to Linux ass a daily driver :)

nov3mb3r, 2020-08-14:
Will give it a try and make new issues for linux this time if I have problem

lvandenb, 2022-02-03:
strange , on my machine , using bootloader on Chameleon , I see no com port, and there is Atxmega128A4U
[image]

david-oswald, 2022-02-03:
@lvandenb that looks like expected behavior: the bootloader is a USB device but does not emulate a COM port. The normal firmware shows then up as COM8 in your case.

lvandenb, 2022-02-03:
yes , it was only to show a normal/working situation, vs the one on top.