emsec/ChameleonMini #293: chameleon doesnˋt activate field

overkillzworld, 2021-01-05:
Hi, recently I bought a used Chameleon Mini (Kasper & Oswald product). 
Flashing the firmware went without problems. 
However, I am not getting an active field in any configuration. 
Do you have any tips on how I could check or fix this? 
What is noticeable is that after starting the Chameleon-Mini, the red LED flashes 2x briefly before the green LED lights up.

Many thanks for your upcoming support.

timokasper, 2021-01-05:
In config=iso14443a_reader, after sending field=1 the 13 MHz RF field should be switched on. 

The 2x ledred flash briefly thing is interesting, I have never seen that...

overkillzworld, 2021-01-06:
> In config=iso14443a_reader, after sending field=1 the 13 MHz RF field should be switched on.
> 
> The 2x ledred flash briefly thing is interesting, I have never seen that...
@timokasper Thank you for your quick support.

My steps:
```field=?
101:OK WITH TEXT
1,0
field=1
100:OK
ledred=field_detected
100:OK
field=0
100:OK
config=mf_classic_1k
100:OK
uid?
101:OK WITH TEXT
FFFFFFFF
ledred=setting_change
100:OK
config=mf_ultralight
100:OK
uid?
101:OK WITH TEXT
FFFFFFFFFFFFFF
```
Now the Chameleon is recognized by phone and reader. Field was active, and tested with ledred=field_detected.
Changed to field=0 and tested other configurations. Now it worked.
Reflashed and tested again. 

Only again after starting the Chameleon-Mini, the red LED flashes 2x briefly before the green LED lights up. 
I wonder what this could be?

https://user-images.githubusercontent.com/77024350/103762643-f41e5d80-5018-11eb-8ed4-89343c9248a4.mp4

overkillzworld, 2021-01-14:
Strangely, the LIR2032 discharges quickly even though the ChameleonMini is set to off. Changed to a new LIR2032 but again dame problem. This seems to have something to do with the flashing of the red LED at the start. It seems as if the ChamelonMini restarts itself again and again. @timokasper Any pointers on how to test it (software side, hardware side)? Reflash again(usb, avrisp)?

david-oswald, 2021-03-31:
Sorry for the delay here - so you're saying the LIR discharges even with the physical switch off? That is really surprising because it should fully disconnect the battery... what behaviour do you get with the battery removed and the Chameleon powered from USB? Same flashing?

david-oswald, 2021-04-01:
Strange, are there any obvious shorts on the board? We always test Chameleons carefully before shipping them, so I'd like to understand what's going on here.

Also, back to your original question: 

> However, I am not getting an active field in any configuration.

So from the above description I take it that the Chameleon emulation is working, but the reader mode is not?

overkillzworld, 2021-04-01:
Its a strange behaviour. Now i activated the field manually and deactivate it again (`field=1` and `field=0`)
Now it works emulating and reading.
Even after a restart it read and emulate.

I couldn't see any shorts on the board.
[image]
[image]

But the battery problem and the flashing on startup still remain.
Should I put this in a new issue?

david-oswald, 2021-04-01:
I think it can stay in this issue - I still don't get the relation between `field` and the emulation working or not... by default the field should be off and hence the emulation should work independent of the field. Can you again explain why you went through your initial steps?

fptrs, 2021-04-01:
Can you check the log and see if the chameleon sends or receives something?

overkillzworld, 2021-04-01:
> I think it can stay in this issue - I still don't get the relation between `field` and the emulation working or not... by default the field should be off and hence the emulation should work independent of the field. Can you again explain why you went through your initial steps?

Initial steps where like @timokasper  proposed:
https://github.com/emsec/ChameleonMini/issues/293#issuecomment-754893169

The state before set field=1 was a fresh flashed Chameleon with no possibility to reading and emulating.
After trying the suggested way. Both work, even after flashing fresh firmware again.
Now default is the field off. I tested it with a 13,56 MHz LED as indicator.
But to reach this state was necessary an initial setting of the field.

overkillzworld, 2021-04-01:
> Can you check the log and see if the chameleon sends or receives something?

In which constellation should I download the log? Boot the Chameleon, read, emulate? Or only the boot process?

fptrs, 2021-04-06:
I would check the log after the boot process. We should see why the Chameleon blinks twice. By default the red led indicates a setting change. You can also check this with the terminal.

overkillzworld, 2021-04-06:
> I would check the log after the boot process. We should see why the Chameleon blinks twice. By default the red led indicates a setting change. You can also check this with the terminal.

@fptrs
How to check in terminal?
`ledred? `shows:
```
ledred?
101:OK WITH TEXT
SETTING_CHANGE
```
Log after boot:
```
Note: If parityBit check failed, '!' is appended to the decoded data and raw data with parity bit is displayed.

00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
```
**OK it seems its behaviour changes if I change slot setting.**
setting=1 many blinks.
setting=8 is not possible.
**But it seems that slot 8 is not accessible.**

Log after trying to change the slot settings and reboot to take a look at the ledred behaviour:

```
Note: If parityBit check failed, '!' is appended to the decoded data and raw data with parity bit is displayed.

00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
13521 ms <+13521 ms>:SETTING SET                  (1   bytes) [2                   ]    
00000 ms <+52015 ms>:BOOT                         (0   bytes) [                    ]    
09436 ms < +9436 ms>:SETTING SET                  (1   bytes) [3                   ]    
00000 ms <+56100 ms>:BOOT                         (0   bytes) [                    ]    
00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
07524 ms < +7524 ms>:SETTING SET                  (1   bytes) [3                   ]    
00000 ms <+58012 ms>:BOOT                         (0   bytes) [                    ]    
06610 ms < +6610 ms>:SETTING SET                  (1   bytes) [4                   ]    
00000 ms <+58926 ms>:BOOT                         (0   bytes) [                    ]    
00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
00000 ms <    +0 ms>:BOOT                         (0   bytes) [                    ]    
11834 ms <+11834 ms>:SETTING SET                  (1   bytes) [5                   ]    
00000 ms <+53702 ms>:BOOT                         (0   bytes) [                    ]    
07418 ms < +7418 ms>:SETTING SET                  (1   bytes) [6                   ]    
00000 ms <+58118 ms>:BOOT                         (0   bytes) [                    ]    
08741 ms < +8741 ms>:SETTING SET                  (1   bytes) [7                   ]    
00000 ms <+56795 ms>:BOOT                         (0   bytes) [                    ]    
10161 ms <+10161 ms>:SETTING SET                  (1   bytes) [8                   ]    
20411 ms <+10250 ms>:SETTING SET                  (1   bytes) [0                   ]    
23667 ms < +3256 ms>:SETTING SET                  (1   bytes) [?                   ]    
28653 ms < +4986 ms>:SETTING SET                  (1   bytes) [1                   ]    
35447 ms < +6794 ms>:SETTING SET                  (1   bytes) [8                   ]    
10149 ms <+40238 ms>:SETTING SET                  (1   bytes) [1                   ]    
00000 ms <+55387 ms>:BOOT                         (0   bytes) [                    ]    
```

fptrs, 2021-04-06:
The Chameleon blinks after boot according to the slot it is in. Setting 8 is currently not accessible due to a wrong definition of FLASH_DATA_SIZE. I will update the Makefile and remove the wrong definition

closed 2021-04-06 by fptrs

libin-ka, 2022-02-02:
If the hardware is made by itself, it will definitely go wrong, because there is an error in the original on the schematic