emsec/ChameleonMini #294: Chameleon RevG flashing firmware not working

fightforge, 2021-02-02:
Hi,
i just cloned this repositoy and then uncomment:

CONFIG_SETTINGS += -DCONFIG_MF_DESFIRE_SUPPORT

in the Makefile in ChameleonMini/Firmware/Chameleon-Mini, and i run make without errors.

Then i put chameleon mini rev G in DFU and write:

sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i

I didn't see errors, but when i disconnect and reconnect chameleon mini revG to linux i cannot get any devices recognized... If i execute dmesg i cannot see nothing. I can see bluetooth device with chameleon app for android but it won't connect.

I want to enable Desfire on chameleon but i think i made a mistake...

What i'm doing wrong? 

Thanks

ceres-c, 2021-02-02:
Sorry, you're talking about the bluetooth version, right? In that case you should ask in [RRG repository](https://github.com/RfidResearchGroup/ChameleonMini), where they maintain code for that device

Anyhow, I fear you hit the space limitations when you built the image with DESFIRE support. Given you flashed the chameleon with the "barebone" avrdude command and did not use `make program`, the file size hasn't been checked prior flashing.
Can you get your PC to recognize the chameleon in bootloader mode? To start it in bootloader mode, plug it in your computer while pressing one of the buttons (I honestly don't remember which one). Make sure the power switch is off during this process

fightforge, 2021-02-03:
Hi,
thanks for reply, yes i restored the original rrg firmware, but i think i need to buy the chameleon mini revE to use desfire mod from this repository... i'm right?

r1ddl3rz, 2021-02-03:
@fightforge Not the RevE you need the RevG. 
Like @ceres-c said -> space limitations - you cannot use the complete default configuration and desfire together.
You could try to use `make desfire` and only use the desfire stuff, iirc.

maxieds, 2021-02-06:
@fightforge 
The firmware is known not to fit in the memory space for the RevE devices. When I wrote the DESFire implementation, I only used the KAOS devices without bluetooth. For some reason, I had some interference with my USB reader and the Chameleon devices with BT running. Your mileage may vary. 

You can tweak the ``Makefile`` settings to include DESFire support. This will mean that most other tag types in the Chameleon firmware will not fit into memory. This is an artifact of the large requirements for the crypto structures with the DESFire implementation. In general, the suggestion from @r1ddl3rz is accurate in that to expect the DESFire support you should build the firmware with ``make desfire``. If you really require DESFire + any other specific tag besides the [standard ones](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Makefile#L7), it may or may not work.

Ch-a-k, 2022-12-19:
Good evening! Bought Chameleon tiny PRO, started installing the update. Apparently he did something wrong. Now tiny itself does not display the active number, 1,2,3,4,5,6 LEDs are lit in pale red. I can't connect via bluetooth to my android or iphone. What to do ?

Ch-a-k, 2022-12-19:
I can't connect it to my PC on windows or Mac too...

ceres-c, 2022-12-19:
@Ch-a-k wrong repository, you should ask at https://github.com/RfidResearchGroup/ChameleonMini

We can’t give you specific info, but I personally had a similar issue and fixed it disconnecting the battery (disassembly required)