emsec/ChameleonMini #295: emulate function problem！

KiraCheng6, 2021-03-02:
i am using the tiny pro(same as the mini revg),i have already upgrade the revg firmware but still having problem.

the situation is:
 i already have a dump and i upload both app and GUI, the dump have change, and i have no idea

the PIC is showing the situation,the left side is the dump i download back from the tiny pro, the right side is the dump i actually upload,

thanks for help!

[image]

KiraCheng6, 2021-03-02:
[image]

david-oswald, 2021-03-02:
Please contact the manufacturer of the Tiny Pro - this repo only supports the original RevG

closed 2021-03-02 by david-oswald