emsec/ChameleonMini #296: Manually insert keys to dump tag data & sniffing com

mgp25, 2021-03-29:
Hello,

I have recovered all the keys from a tag with proxmark using the nested attack. I want to use ChameleonMini to dump contents of the tag after being used in a remote place, in order to do this, I would need to either crack the Mifare keys using the chameleon android app, or the way I am trying to do, which is manually placing the keys that I already have into the android device and be able to read the tag contents in any moment that I want.

Where should be this file placed and which format does it has? If its the same as proxmark, I would only need the path.

On a side question: I have read the documentation but I am not able to understand yet how the sniffing communication works. As I understood, just by placing the ChameleonMini between the reader and the tag. What I don't know is how to operate it with the android app or if is not required at all. I configure `slot 2`for `ISO14443A_SNIFF`. What else should I need to do to capture the communication between the reader and the tag.

Best regards.

david-oswald, 2021-03-31:
Hi,

So do you want to emulate the card that you read (and have the keys for) using the Chameleon? In that case you can simply `upload` a dump (mfd file format which is just a binary, I think this is what the Proxmark outputs as well) to the Chameleon, then use it at the reader, `download` and you should have the changes.

As for the sniffing, if it's related to the android app then I suggest to discuss that in the repo/forum of the app.

mgp25, 2021-04-01:
Hello @david-oswald,

Thanks for answering! That will definitely work. I am going to test it in a real scenario and will check if everything goes as planned. I will try to give it a shot to the sniffing part. I guess it will be stored in the logs maybe. I will update this as soon as I test it.

Best regards

mgp25, 2021-04-04:
@david-oswald Everything worked awesome! Thanks again for your help, as per how the sniffing works with the mobile app, will ask in the mobile app repo and update this issue when I have an answer for that.

Best regards

closed 2021-04-04 by mgp25