emsec/ChameleonMini #301: Battery draining

overkillzworld, 2021-06-24:
What is the best way to find out why my ChameleonMini always drains the battery even though it is turned off?
Already flashed latest firmware.
Tested different new LIR2032.
Fully charged via USB. 
Power switch is turned off.
After one day it is empty.
Some advice on how to proceed would help me.

david-oswald, 2021-06-24:
Hi, 

As you can see in the schematics, if the power switch is off the battery is only connected to the charger IC:

[image]

So I would check how that IC looks / if it gets hot or similar. Another possible issue might be that C4 (parallel to the battery) has failed and conducts (a bit) to ground, do you have a multimeter to check that?

Overall, whether you can repair this yourself would of course depend on your electronics experience.

overkillzworld, 2021-06-25:
Yes I have a multimeter here. However, I am a developer and not an electronics expert :) What is the best way to measure such a small MLCC? Sorry for the stupid question, but only so I can learn something about it.

david-oswald, 2021-06-25:
You can take out the battery and then measure the resistance between battery + and - 

This should be high impedance / constantly increase, otherwise there might be an issue with the charging chip or the cap.

overkillzworld, 2021-06-25:
@david-oswald Thanks for the fast reply.
So it is correct:
Position:
[image]
Measure:
[image]
[image]
[image]

david-oswald, 2021-06-25:
Yes, so this looks like there's no low impedance in the cap or charging IC - just to make sure, does it go up further than 3.8 M? (this will depend on the multimeter though)

Also, can you check/read the value of R4 on your device?

Finally, have you measured the battery voltage when "fully charged"?

timokasper, 2021-06-26:
The LIR2032 has 40mAh with 3.6-4.2 Volts. Current=Voltage/Resistance, thus there is a current of about 1 micro Ampere. Thus it takes 40.000 hours / 5 years to drain a full battery ...

so your problem must be elsewhere. 

Are you sure you are using an LIR2032 not a CR2032...     You can measure the voltage at the battery. 4.2 is full, 3.6 is empty, <=3.3 is either a wrong battery (CR2032) or a dead LIR cell

overkillzworld, 2021-06-28:
> Yes, so this looks like there's no low impedance in the cap or charging IC - just to make sure, does it go up further than 3.8 M? (this will depend on the multimeter though)
>
> Also, can you [...]

It stops at 3.85 M Or like seen in the picture 3.89 M.

overkillzworld, 2021-06-28:
> The LIR2032 has 40mAh with 3.6-4.2 Volts. Current=Voltage/Resistance, thus there is a current of about 1 micro Ampere. Thus it takes 40.000 hours / 5 years to drain a full battery ...
>
> so your [...]

I am sure it is a LIR2032. Ordered a pack to test if the old LIR was maybe dead. Measured the LIR2032 full charged it has 4.19 V

overkillzworld, 2021-06-28:
Now that the battery is fully charged, I'll see how long it takes to completely discharge it. Last time it was 1-2 days.

overkillzworld, 2021-07-02:
I have an Idea what could cause the battery problem, but I wanted to look at the schematic first. Short question offtopic: Which program is best to view the Schematic [.sch] files?  If I try it in KiCad it looks not correct
[image]

david-oswald, 2021-07-02:
It's in Eagle, but there is also a PDF in the folder so that you don't have to install that just to check the schematic.

overkillzworld, 2021-07-02:
Ok. Thanks. I'll take a closer look :)

overkillzworld, 2021-07-06:
Could it have been a short on the board between C15 and IC2 (or C4 and IC2)?  There the solder joints were a bit close together or slightly connected. I have corrected this and now it looks better since a few days. I'll keep watching it.