emsec/ChameleonMini #307: Links to Firmware on Getting Started page lead to 404 error

kampfhamster309, 2021-11-20:
I just got my chameleon and tried to follow the getting started page. Unfortunately, the links on it seem to be broken/old. If i try to access 

https://cdn.statically.io/gh/emsec/ChameleonMini/master/Firmware/Chameleon-Mini/Latest/Chameleon-Mini.hex

or

https://cdn.statically.io/gh/emsec/ChameleonMini/master/Firmware/Chameleon-Mini/Latest/Chameleon-Mini.eep

to grab the new Firmware -> all I get is a 404 Error. It would be great if you could update those links or at least remove them.

fptrs, 2021-11-23:
Hi @kampfhamster309,
I checked the getting [started page](https://cdn.statically.io/gh/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html) and it correctly refers to the [release section](https://github.com/emsec/ChameleonMini/releases). Can you tell me where you found the links?

kampfhamster309, 2021-11-23:
It looks to me like there is an older version of the getting started page hanging around... 

If I look at the README.md in this very repo

[image]

and click on the "getting started" link, I land on another version than you linked above.

There, the step 2 part looks like this:

[image]

The release section is not mentioned there, and the two links to the *.hex and *.eep files are the ones I posted above, which lead to the 404 error. 

I ended up just building the files myself from source, but maybe some more inexperienced people would benefit if you update the README.md to the correct getting started section.

fptrs, 2021-11-23:
Thank you @kampfhamster309 for pointing this out. I only checked the Wiki page before

commit 61b547a1 (2021-11-23): Update README.md

update link fix #307
closed 2021-11-23 by fptrs