emsec/ChameleonMini #308: Can't toggle slots

stematon, 2021-12-11:
Hi!
I am running the Version 61b547a on a ChameleonTiny. If I understand this correctly this firmware is compatible with the tiny, too? 
At some very early point of using the device I can't change the slots with the buttons anymore. The LED of slot 2 is always on as long as I don't set the LED to off. 
Even with a new firmware flash I could not get it back running. Any ideas?

david-oswald, 2021-12-13:
Hi @stematon we are not supporting the ChameleonTiny in this repo - they might well have made it compatible with our firmware, but any bug/issue with their device should go to their repo, sorry.

closed 2021-12-13 by david-oswald

david-oswald, 2021-12-14:
I guess you can check here: https://github.com/RfidResearchGroup/ChameleonMini/issues or the RRG discord, that is also quite active.