emsec/ChameleonMini #310: DESFire testing equipment for development (input needed)

maxieds, 2021-12-19:
Hello all. I mentioned in #302 that I have received funding from GT to work more on the DESFire emulation support this spring. I am looking to purchase more hardware (readers, proxmark, etc.) to help with testing and development. I already have a ACR-122 reader that I know works with my Mac. Can you all please suggest additional good hardware to enumerate and collect for my work on this project? I want the end result to be as seamlessly debugged and feature full as possible. It's also important to note that I am on a budget, so suggestions towards working within that (i.e., hardware that gets the "best bang for the buck") are appreciated here as well.

lvandenb, 2021-12-20:
For me the 14443 part is the more difficult part.
Debugging will require a  chameleon or PC532 , or proxmark ..

Personally I use omnikey 5022CL, 5127CK. They are handling the 14443A automatically, but so is your ACR-122.
I also ordered a TWN4 reader from elatec. The latter contains a SDK with a higher C Api ( which is free) . The 14443A is handled in background, so we can concentrate on the Desfire EV1 itself
Also nice is the free “NXP TagInfo” App on Android. It gives a quick overview of the Desfire EV1 Card, conform the NXP standards.

I wrote the client side, using PC/SC,  in a badge rollout application.
some apps on the card containing encrypted files, some public apps and one NFC app. The apps use AES, the PICC  uses 2KDes for authentication.

It would be nice to use the Chameleon as a real card..

maxieds, 2021-12-22:
@ lvandenb 
Thanks for the helpful suggestions. I am going to order a PM3, the Omnikey 5022CL, and most likely a TWN4 to do testing. When it all arrives, I should be able to give better responses on the Omnikey reader problems you had in #302 as I will be able to debug locally. 

I just wanted to mention that I might have less time for this project in the Spring than anticipated. I am planning to graduate and defend this Spring (or possibly worst case in the Summer). My GT math advisor issued me a "Get thy thesis to me [NOW!]" mandate this week when I asked him to fund the hardware for the DESFire project. That's my top priority. Nonetheless, I am supposed to be working around 8 hours a week on DESFire (on paper)...

maxieds, 2021-12-23:
@Ivandenb
I am looking at ordering [this version of the Elastec device](https://www.tme.com/us/en-us/details/t4bt-fb2bel4/rfid-modules-and-readers/elatec/twn4-multitech-2-lf-125khz/?brutto=1&currency=USD). It is listed as operating at 125KHz. Is the Chameleon actually capable of transmitting over that frequency?

r1ddl3rz, 2021-12-23:
> @Ivandenb
> I am looking at ordering [this version of the Elastec [...]

No the chameleon is only HF (13.56 MHz)

lvandenb, 2021-12-25:
there is a slightly more expensive version, doing 13MHz mifare and desfire. also having 2 sam slots.
The one you show is the stripped down version.
I ll get mine at beginning jan 22.  I ll let you know the Desfire ev1 test results. (client side to real nxp desfire ev1)

lvandenb, 2021-12-25:
Ik guess the crc is missing in the SAK.
The omnikey reader is strict..

Met vriendelijke groeten,
Luc Vandenbroucke

lvandenb, 2021-12-25:
It is the slightly more expensive  version. With 13.xxMhz and 2 Sam slots
The SDK is easily requested from elatec.
I' ll let you know feedback of our Desfire ev1 testing in januari

Met vriendelijke groeten,
Luc Vandenbroucke

maxieds, 2022-01-18:
I think we can safely close this issue now. Thanks for the helpful suggestions. The equipment I purchased helped in making a (to be filed sometime soon) pull request to resolve #302 and #313. 😸

lvandenb, 2022-01-27:
I just tested the TWN4 multitech. 

It is a nice platform for developing ( also with gcc compiler), and there is a director app, to test most RF card functions. (neeeds free sdk)
Like scriptor advanced, allowing  desfire EV1 (and other technology) authentication, encrypted or cmac'ed files, ...
but also 8716 apdu. also crypto , only lacking cmac for the moment.
it works as CDC (com port simul), HID keyboard , pc/sc, just choose the appropriate firmware.
disadvantage could be : the tools delivered are windows based.  but the commands can be send from any system, using usb serial.

maxieds, 2022-02-02:
I am trying to acquire a Proxmark3 RDV4 device [here](https://hackerwarehouse.com/product/proxmark3-rdv4-kit/). Which of the optional add-ons will I need for Chameleon DESFire testing? Both antennas, or just one?

fptrs, 2022-02-03:
You won't need any additional antennas. The device already compatible with 13.56MHz.