emsec/ChameleonMini #317: Create a project to keep track of misc DESFire support TODO tasks?

maxieds, 2022-01-31:
I do not mean to be any more demanding or more of a PITA on this forum than usual, but this seems like a reasonable thing to request. The discussion in #313 is getting cluttered with [posts noting status code return inconsistencies](https://github.com/emsec/ChameleonMini/issues/313#issuecomment-1019252143) and a [moving discussion on authentication command variants](https://github.com/emsec/ChameleonMini/issues/313#issuecomment-1024400639). In principle, it makes sense to open a new issue for each minor problem that comes up with the DESFire emulation support. That will probably never actually happen. So I move to create a TODO-list type project for the DESFire support where things can get categorized and logged in a sensibly organized manner. This will help me (right now) and others that want to contribute keep track of what still needs attention. For example, I have [something like this](https://github.com/maxieds/ChameleonMiniLiveDebugger/projects/3) on my Chameleon related Droid phone app repository. Any input from the maintainers on this topic?

lvandenb, 2022-01-31:
I agree it needs some structure now. 
This scripting will be nice to run all the testing. 
Also on the side of the PCD writer we make testing scripts . (Each script will make a some desfire config.)

starting like a 1) simple app, containg a read write file , 2: a NFC tag on desfire ev1 ,  3: an app with 2 AES  keys, + encrypted file, 4: a longer file ( > 65 bytes)
When the authentication random number generator is set to a known value at both sides,  even the basic scriptor can be used, because all commands will look he same every time. 
( The scripts can also be automated on a twn4 multitech (HF version). ( one of the firmware versions allows commands be sent over usb  rs232 emul (cdc))

so each open issue can be identified with project x, step y

maxieds, 2022-01-31:
My question is whether someone (me perhaps) can be the delegate to organize this project? I am sure there are settings available to make only some options (like editing / modifying the projects) available without having to enable full permissions on the repo.

david-oswald, 2022-01-31:
I created this: https://github.com/emsec/ChameleonMini/projects/1

Can you see that @maxieds?

I will see if I can give permission to the project to others...

maxieds, 2022-01-31:
@david-oswald 
Yes, I can see it. Again, I get why you will be touchy about letting someone else have moderator access to parts of this repo. It would be nice if I could add some cards to the project and start categorizing some stuff that I can already anticipate coming up with the authenticate commands discussion in #313. For example, this is what I can see in my project:
[image]
And this is what I can currently see on the project you just created:
[image]

Some ideas for subprojects (just to get this off my mind this morning):
* Cryptographic exchanges and authentication modes (this is going to be involved ...)
* ISO7816 and ISO14443-3/4 testing for compliance with real DESFire tags
* The scripting topics that @lvandenb mentioned to setup stock cards and help facilitate testing en masse
* Possibly create better docs for all the things that can be done with the layers of this implementation. The one I added with the first round of PRs about 1.5 years ago is messy and writen for hackers and dataminers only.

david-oswald, 2022-01-31:
Hi @maxieds - I could not find any option to give access to the project part only for other users... I think there is some way by using the organisation, but Github is fairly confusing about this (alone that there are two menu entries called "Projects" but with different icons and also both projects for a single repo and then for the org...).

I also created this now, maybe that's better (I tried to give you r/w access to that): https://github.com/orgs/emsec/projects/3

closed 2022-06-16 by maxieds