emsec/ChameleonMini #324: Where to download the firmware binaries built with the emsec/ChameleonMini actions scripts?

maxieds, 2022-07-27:
One of the recent pull requests I submitted had [custom ``make`` targets](https://github.com/emsec/ChameleonMini/blob/master/Doc/BuilingFirmwareBinariesFromSource.md#customizing-the-build) to build the firmware. @david-oswald suggested to keep things simple for new users and only provide a few of the most common builds via the automatic GitHub build scripts for this project. I still cannot find where the official firmware builds are located. I posted up-to-date binaries for the DESFire targets in #321 until the PR #323 is merged.

I searched around and found links to using the GitHub REST API. The pointers [on this page](https://docs.github.com/en/rest/actions/artifacts#download-an-artifact) suggest something like:
```bash
$ curl -H "Accept: application/vnd.github+json" https://api.github.com/repos/emsec/ChameleonMini/actions | head -n 32
$ curl -H "Accept: application/vnd.github+json" https://api.github.com/repos/emsec/ChameleonMini/actions
```
Trying to download via the command line gives ``ERROR 403: Forbidden``. using ``wget https://api.github.com/repos/emsec/ChameleonMini/actions/artifacts/309513562/zip``. Navigating to the download link by web browser yields an error about [invalid access tokens](https://docs.github.com/en/rest/actions/artifacts). The required access token for this public repository must have the [``public_repo`` actions scope](https://docs.github.com/en/rest/actions/artifacts). I have a personal access token with all of the relevant scopes applied that I can see and the download is still forbidden:
```bash
$ curl -H "Accept: application/vnd.github+json" -H "Authorization: tokenghp_<REST-OF-TOKEN-STRING>" https://api.github.com/repos/emsec/ChameleonMini/actions/artifacts/309513562/zip 
{
  "message": "You must have the actions scope to download artifacts.",
  "documentation_url": "https://docs.github.com/rest/reference/actions#download-an-artifact"
}
```
Is it possible that there is a configuration setting on the admin side of the repository permissions that is the problem? Has anyone else without admin authentication on this repository had luck downloading these files? I cannot figure it out.

fptrs, 2022-07-27:
The firmware binaries are available on the [release page](https://github.com/emsec/ChameleonMini/releases).

maxieds, 2022-07-28:
Check! 👍🏻

closed 2022-07-28 by maxieds