emsec/ChameleonMini #325: Lost ability to store card data to FRAM in all releases after Jul 08, 2022

Heavy-Horses, 2022-09-06:
I've lost the ability to store card data to FRAM in all releases after Jul 08, 2022. I don't have this issue the June 29, 2022 release.

fptrs, 2022-09-06:
Thanks for the hint @Heavy-Horses. I already found the origin of this issue.
Since PR #319 all memory is cleared when a new configuration is set:
```
void ConfigurationSetById(ConfigurationEnum Configuration, bool appInitRunOnce) {
[...]
/* Blank memory scape slate for the newly set configuration */
MemoryClear();
```

@maxieds can we revert this without affecting your code?

maxieds, 2022-09-16:
@fptrs 
Sorry. I have been out sick for the last couple of weeks since moving to the desert. I still haven't setup my home machines, so I will be unable to test what the change does. 

Some comments based on the [code cited above](https://github.com/emsec/ChameleonMini/blob/bc333f9dd1df72d1becfc1186acfac9dc1c8f094/Firmware/Chameleon-Mini/Configuration.c#L524) and the corresponding [memory code in the firmware](https://github.com/emsec/ChameleonMini/blob/bc333f9dd1df72d1becfc1186acfac9dc1c8f094/Firmware/Chameleon-Mini/Memory.c#L391):
* You might still need the call to ``MemoryRecall();`` [from here](https://github.com/emsec/ChameleonMini/blob/bc333f9dd1df72d1becfc1186acfac9dc1c8f094/Firmware/Chameleon-Mini/Memory.c#L393). 
* IIRC, the intent of that code in ``ConfigurationSetById`` was to initialize a new tag configuration to have blank data (or restore the previously saved state). But then again, this unintentionally introduced the issue at hand by me only testing DESFire with ``LOGMODE=LIVE``.
* If you want to make sure the change does not affect the DESFire firmware builds, I think all you need to check is that a previous configuration set by an initial ``CONFIG=MF_DESFIRE`` gets restored correctly (UID and maybe structures depending on how thoroughly you want to test). This was kind of tricky to ensure. That's all I would really be worried about.

commit 9fbb7fbb (2022-09-21): fix #325

* remove MemoryClear from ConfigurationSetById
closed 2022-09-21 by fptrs