emsec/ChameleonMini #331: DESFire: Unable to auth to apps unles authed to the PICC mater app first

tomaspre, 2022-12-06:
When running the DESFire config, I'm not able to authenticate to any app (except the PICC master app) unless I authenticate to the PICC master app first. Therefore, a card can be set up without any problems, but when the authentication state is reset, access to all apps (except the master one) is lost, even if the reader authenticates with the correct key for that particular app.

Example output from `pm3` of what does not work (auth to a previously created application):
```
[usb] pm3 --> hf gallagher reader --apdu 
[+] Setting ISODEP -> inactive                                                                                   
[+] Setting ISODEP -> inactive                                                                                   
[+] >>>> 90 5A 00 00 03 F4 81 2F 00                                                                              
[+] Setting ISODEP -> inactive                                                                                   
[+] Setting ISODEP -> NFC-A                                                                                      
[+] <<<< 91 00                                                                                                   
[+] >>>> 90 BD 00 00 07 00 00 00 00 24 00 00 00                                                                  
[+] <<<< 02 03 F8 20 81 F4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 91 00 
[+] Setting ISODEP -> inactive                                                                                   
[+] >>>> 90 5A 00 00 03 F4 81 20 00                                                                              
[+] Setting ISODEP -> inactive                                                                                   
[+] Setting ISODEP -> NFC-A                                                                                      
[+] <<<< 91 00                                                                                                                                               
[+] >>>> 90 AA 00 00 01 00 00                                                                                    
[+] <<<< 91 9D                                                                                                   
[!!] 🚨 APDU(90aa) ERROR: [0x919D] Not allow the requested command                                               
[!!] 🚨 Desfire authenticate error. Result: [1] Sending auth command failed                                      
[!!] 🚨 Failed selecting/authenticating to AID 2081F4                                                            
[!!] 🚨 Failed reading card application credentials
```

This could potentially be fixed by removing the following lines from `DESFire Instructions.c` (around line 2014, in function `DesfireCmdAuthenticateAES1` ):
```
if (!Authenticated && !AuthenticatedWithPICCMasterKey && SelectedApp.Slot != DESFIRE_PICC_APP_SLOT) {
        Buffer[0] = STATUS_PERMISSION_DENIED;
        return DESFIRE_STATUS_RESPONSE_SIZE;
    }
```

Is this the correct approach? According to my testing, everything works as expected after commenting out these lines. Does anyone have an opinion on this? I can submit a PR if necessary.

maxieds, 2023-10-31:
@tomaspre
Do you have a spare Chameleon Mini device for me to debug with? That would facilitate bug fixes.

maxieds, 2023-11-01:
I think that the DESFire datasheets I looked at when developing the DESFire emulation (see [links here](https://github.com/emsec/ChameleonMini/blob/master/Doc/DESFireSupportReadme.md#links-to-public-datasheets-and-online-specs)) specify that if there is no existing state (or state has reset), the first thing is to authenticate to the PICC master application. The problem is probably that the first (legacy) round of authentication needs to use DES or 3DES as an encryption scheme. It seems reasonable to let the scheme be AES128 if the tag is configured that way.