emsec/ChameleonMini #333: Mifare Ultralight EV1 Emulation Support

Suhas-13, 2023-01-17:
Hi there,

I was wondering if Mifare Ultralight EV1 card emulation is properly supported on this firmware. I attempted to clone a 16 page Mifare Ultralight EV1 NXP card, but it didn't seem to work with the reader. Is this not yet supported?

ceres-c, 2023-01-17:
I haven’t personally used Mifare Ultralight EV1 emulation, but it is mentioned in the code for the Ultralight application.
Have you checked the logs? When does the chameleon stop responding?

Suhas-13, 2023-01-17:
The reader blinks red when I use the Chameleon Mini. It doesn't stop responding.

Suhas-13, 2023-01-17:
I think the issue is that the Mifare Ultralight EV1 is 20 pages, but the card I have is 16 pages.

ceres-c, 2023-01-17:
If the reader blinks red, it means it does not like something in the data it receives from the card. Check the logs and, most likely, the last command in the log issued by the reader will be the one yielding the rejected answer