emsec/ChameleonMini #334: cannot access device at all

dxvdmj, 2023-03-04:
So I bought one of these guys a long time ago in the summer, chameleon tiny pro with BT, problem being is that I believe I re-mapped the on button and somehow disabled BT function so I cannot turn it on with phone or by its self...rendering it unusable. Wondering if there are any ways to just completely reset the device or get back to factory settings with some kind of software?

dxvdmj, 2023-03-04:
^ when plugged into computer it charges and I can use the buttons but there's no way to turn BT back on to properly use it with my phone is the issue

david-oswald, 2023-03-04:
Hi, we don't support Chameleon Tiny in this project; please ask the manufacturer. We only deal with the original Chameleon RevG here.

closed 2023-03-04 by david-oswald