emsec/ChameleonMini #336: Issues with Mifare DES-Fire Emulation

flo-vienna, 2023-08-23:
Used Device: ChameleonMini RevG
Used Firmware: DESFire-only Firmware Build 25e6545f9dd3cbf64ca6745db26663204684587a

ATS as reported by proxmark3:
[+] ATS: 06 75 77 81 02 80 [ F0 00 ]

Steps to reproduce:

I tried to follow the instruction on this page:
https://github.com/emsec/ChameleonMini/blob/master/Doc/DESFireSupportReadme.md

Executing the following command(s) does indeed work:
CONFIG=MF_DESFIRE
or
CONFIG=MF_DESFIRE_2KEV1

Any subsequent requests like 
DF_SETHDR=ATS 067577810280

lead to an 
201:INVALID COMMAND USAGE0 
error no matter what I try, the same goes for 
DF_SETHDR? or DF_SETHDR=?

On a side-note, the UID is set via an ISO14443A command as this is not mentioned on the page but somehow in the code ?

tomaspre, 2023-08-24:
Hi Flo,

I can confirm that I can reproduce the issues with the DF_SETHDR terminal command. I'll debug this and get back to you ASAP.

The command for changing UID is not DESFire-specific; therefore, that's not in the DESFire-specific readme. You can use the usual ´UID=´ and ´UID?´ commands.

tomaspre, 2023-08-25:
Hi @flo-vienna,

I pushed a fix into the fix336 branch. The builds are below. You should now be able to set the ATS to any value as long as you set the first ATS byte to be the ATS size (as per DESFire specs). The updated DESFire-specific readme is here:
https://github.com/emsec/ChameleonMini/blob/fix336/Doc/DESFireSupportReadme.md

Please let me know if these fixes work for you so I can merge them into master.

[Chameleon-Mini-CustomBuild_DESFire-fix336.zip](https://github.com/emsec/ChameleonMini/files/12439308/Chameleon-Mini-CustomBuild_DESFire-fix336.zip)

commit 724ce5b9 (2023-08-29): Fix #336, DESFire ATS can be set to any size from 3 to 20