emsec/ChameleonMini #339: No communication between old NFC reader and ChameleonTiny Pro

HainBuche, 2023-10-31:
Hi,

I have an old reader (pre 2010) with unknown specifications that was used for 1k MIFARE classic cards.
It can still read the old cards successfully. These cards can also be read on a newer reader (such as ACR-122) and can also be read by the ChameleonTiny and further be successfully copied and emulated and then read by the newer reader. However, emulated cards are not recognized by the old reader, and even more important the ChameleonTiny does not seem to recognize any near field signal from the old reader, as the RF-LED does not light up in neither mode (emulate, read, sniff) and no Log is created.
What might be the reason ? Can there be timing issues ? Other hardware issues ?
Hope you can give me some insights, thanks !

Best regards

ceres-c, 2023-10-31:
Hello, this is the repo for the ChameleonMini, for the ChameleonTiny you should ask [here](https://github.com/RfidResearchGroup/ChameleonMini)

My take on your issue is that the antenna is too small and the reader don’t like it. But I am just guessing

HainBuche, 2023-10-31:
Hey, I actually posted there first, but was unsure.
But thanks for your quick response :-)
If it would be indeed the antenna, how would you approach to fix it, attach some wire?
I had tried moving the ChameleonTiny to different positions and distances, but it did not help.
Thank you for your time!

ceres-c, 2023-10-31:
The problem with the ChameleonTiny is indeed that you can't (AFAIK) attach an external antenna. You can do that on the ChameleonMini, quite easily so with an SMA connector actually, but the other device is more end-user oriented rather than research so IIRC some feature such as the SMA have been dropped in favor of smaller size and sleek looks.
If you were able to, you could indeed try to use an antenna with a different shape/size. There are some tools online to calculate NFC antennas geometries, as well as vendor specific tools: search ST and TI websites.

If I had to guess, I'd say the reader you're dealing with is a bit "deaf" (read: bad antenna design) and doesn't detect the small load induced in its field by the tiny antenna in the ChameleonTiny

HainBuche, 2023-11-01:
Thanks a lot, I'll look into that and see whats possible for me. What puzzles me though is that apparently even when placed between the old reader and a card while they are communicating, no signal is detected by the ChameleonTiny. Might that still be explainable by the antenna issue ?

ceres-c, 2023-11-01:
It is indeed weird, but IIRC the field detector in the ChameleonTiny is essentially just an led connected to the antenna. If indeed the cham/reader antenna coupling is bad, that would explain both because:
1) There will not be enough energy on the Chameleon's side to power up the led
2) It will not be detected by the reader because there is no load

If you think about it, it's like moving a random piece of metal within the reader field

HainBuche, 2023-11-01:
Oh okay that sounds quite plausible. Guess I will have to putter around a little then, hope I don't ruin it.
Thanks again for your quick response.

closed 2023-11-02 by HainBuche