emsec/ChameleonMini #341: UID set

sina6611, 2023-11-10:
Hi everyone,

I want to set a new UID in the Chameleon Mini and I have used the command UID=<UID>. For example, I have set UID=<1234567989ABCDEF>, but the Chameleon Mini answers INVALID parameter. Does anyone know how I can set a new UID on it?

spankywetfish, 2023-11-10:
You don’t need the `<>`, they are there to indicate that a parameter is required.
E.g. `UID=12345`

r1ddl3rz, 2023-11-10:
> Hi everyone,
> 
> I want to set a new UID in the Chameleon Mini and I have used the command UID=<UID>. For example, I have set UID=<1234567989ABCDEF>, but the Chameleon Mini answers INVALID parameter. Does anyone know how I can set a new UID on it?

And you need to have a valid length for the UID. 
For example cfg=MF_CLASSIC_1K
Correct: UID=11223344 
Invalid Parameter: UID=1122334455