emsec/ChameleonMini #342: MIFARE DESFire EV1 8K

sina6611, 2023-11-14:
Hi Every one
I am trying to set the configuration of the Chameleon Mini to MIFARE DESFire EV1 8K, but I receive a message saying "invalid parameter." I have loaded the latest firmware version, but this configuration does not have a parameter for MIFARE DESFire EV1 8K. Does anyone know how I can fix this?

spankywetfish, 2023-11-14:
Given the following code from `Configuration.c` ...

```
#ifdef CONFIG_MF_DESFIRE_SUPPORT
    { .Id = CONFIG_MF_DESFIRE,                  .Text = "MF_DESFIRE" },
    { .Id = CONFIG_MF_DESFIRE_2KEV1,            .Text = "MF_DESFIRE_2KEV1" },
    { .Id = CONFIG_MF_DESFIRE_4KEV1,            .Text = "MF_DESFIRE_4KEV1" },
    { .Id = CONFIG_MF_DESFIRE_4KEV2,            .Text = "MF_DESFIRE_4KEV2" },
#endif
```
I think it's reasonable to assume that "MIFARE DESFire EV1 8K" isn't supported.

sina6611, 2023-11-14:
Should I give the code in my terminal?

closed 2023-11-30 by tomaspre